raspberrypi/pico-sdk #3: N_GPIOS was not declared in this scope, unless gpio.h is explicitly included

Gadgetoid, 2021-01-22:
I'm not sure this qualifies as an "issue" since I fixed it with: https://github.com/pimoroni/pimoroni-pico/commit/6a13dc5c4b8535df5c7e075cb20728b4be831f53#diff-ddd42d56faea5530bfb4a0200db4bbecaf574481ae9cfa7d66e8b296783ce838

However if don't explicitly:

```c++
#include "hardware/gpio.h"
```

before:

```c++
#include "hardware/pwm.h"
```

Then compiling will fail with:

```
[build] /home/phil/Development/pico/pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h: In function 'uint pwm_gpio_to_channel(uint)':
[build] /home/phil/Development/pico/pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h:86:33: error: 'N_GPIOS' was not declared in this scope
[build]    86 |     valid_params_if(PWM, gpio < N_GPIOS);
[build]       |                                 ^~~~~~~
[build] /home/phil/Development/pico/pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h:86:5: note: in expansion of macro 'valid_params_if'
[build]    86 |     valid_params_if(PWM, gpio < N_GPIOS);
[build]       |     ^~~~~~~~~~~~~~~
[build] /home/phil/Development/pico/pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h:86:33: note: the macro 'N_GPIOS' had not yet been defined
[build]    86 |     valid_params_if(PWM, gpio < N_GPIOS);
[build]       |                                 ^~~~~~~
[build] /home/phil/Development/pico/pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h:86:5: note: in expansion of macro 'valid_params_if'
[build]    86 |     valid_params_if(PWM, gpio < N_GPIOS);
[build]       |     ^~~~~~~~~~~~~~~
[build] In file included from /home/phil/Development/pico/pico-sdk/src/rp2_common/hardware_adc/include/hardware/adc.h:12,
[build]                  from /home/phil/Development/pico/pimoroni-pico/libraries/pico_explorer/pico_explorer.cpp:5:
[build] /home/phil/Development/pico/pico-sdk/src/rp2_common/hardware_gpio/include/hardware/gpio.h:136: note: it was later defined here
[build]   136 | #define N_GPIOS 30
```

It does strike me that `pwm.h` should do this implicitly if it cannot build without that definition. But I could be doing something silly?

kilograham, 2021-01-22:
yes that is a bug - I think there are a few like that lurking

thanks!

kilograham, 2021-01-31:
merged into develop

closed 2021-01-31 by kilograham