raspberrypi/pico-sdk #5: Unable to build in Visual Studio Code

Bazz-Man, 2021-01-23:
I followed the instructions in the PDF and successfully setup Windows to build from the command line.  However when trying to build from VSC it reports the following error:

[main] Building folder: pico-examples 
[build] Starting build
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.EXE" --build c:/Users/barry/Downloads/pico-examples/build --config Debug --target all -- -j 8
[build] ninja: error: FindFirstFileExA(c/:/users/barry/downloads/pico-sdk/src/rp2_common/boot_stage2): The filename, directory name, or volume label syntax is incorrect.
[build] 
[build] Build finished with exit code 1

If I paste the same command line into a VS cmd box it reports the same issue.  This also happens if I select build all or a specific code set e.g. blink

kilograham, 2021-01-23:
I think there is a problem with the ninja CMake generator and pico-examples, can you try setting "NMake Makefiles" as the preferred/default build generator in CMake plugin settings?

realjanek, 2021-01-24:
'nmake' is not recognized as an internal or external command,
operable program or batch file.

I follow the instructions in pdf running on Windows. All good until trying to run nmake.

D:\2021\RpiPico\pico-examples\build>cmake -G "NMake Makefiles" ..
Using PICO_SDK_PATH from environment ('..\..\pico-sdk')
Pico SDK is located at D:/2021/RpiPico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gcc.exe
Defaulting PICO target board to pico since not specified.
Using board configuration from D:/2021/RpiPico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: C:/Python38/python.exe (found version "3.8.7") found components: Interpreter
TinyUSB available at D:/2021/RpiPico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
ELF2UF2 will need to be built
PIOASM will need to be built
-- Configuring done
-- Generating done
-- Build files have been written to: D:/2021/RpiPico/pico-examples/build

D:\2021\RpiPico\pico-examples\build>nmake
'nmake' is not recognized as an internal or external command,
operable program or batch file.

kilograham, 2021-01-24:
you need to run it from a visual studio developer command prompt (or whatever it is called)

realjanek, 2021-01-24:
> 
> 
> you need to run it from a visual studio developer command prompt (or whatever it is called)

Thanks a lot. It works. You save me a lot of time and headaches. I still have to figure out how to use VSCode. Is it possible to use J-Link for debugging on Windows?

Bazz-Man, 2021-01-24:
> I think there is a problem with the ninja CMake generator and pico-examples, can you try setting "NMake Makefiles" as the preferred/default build generator in CMake plugin settings?

Excellent, that fixed it.  So for everyone else, you need to add the "NMake Makefiles" to the Settings tab of the extension under the entry labelled: 
Cmake: Generator
The CMake generator to use 

Should look like this
[image]

kilograham, 2021-01-25:
Also if you are using VS Code on something other than Windows you likely need to change this to "Unix Makfiles"

aallan, 2021-01-26:
I've now updated the documentation to reflect this, will be available in the next docs release expected today.

closed 2021-01-26 by aallan

vishnumaiea, 2021-01-27:
> I've now updated the documentation to reflect this, will be available in the next docs release expected today.

You may also have to specify **not to skip** installing the **Windows 10 SDK** from VS build tool installation step. Because not installing the SDK will cause NMAKE to fail. Installing the Windows 10 SDK solved an error I encountered later.

lurch, 2021-03-29:
Just for the sake of completeness (not that it really matters, because our instructions say to use `nmake`), but it turns out that the problem with `ninja` is that GCC outputs dependency-information in a slightly-incorrect format that `ninja` can't understand.
https://gitlab.kitware.com/cmake/cmake/-/issues/21997#note_929924

aallan, 2021-03-30:
I'll add a note to the Getting Started book to this effect.

eriklundh, 2021-05-17:
The fix for  [ninja-build/ninja#1774](https://github.com/ninja-build/ninja/pull/1774) included in ninja release 1.10.1 and later