raspberrypi/pico-sdk #11: ninja: error: FindFirstFileExA

james-rich, 2021-01-24:
Setup
- Windows 10
- VS Code

Using VS Code I can build the project but after every build I have to manualy remove the content of ".\build\" if I do not I get the following error.

`[main] Building folder: test 
[build] Starting build
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.EXE" --build c:/Users/{USERNAME}/Documents/pico/test/build --config Debug --target all -- -j 14
[build] ninja: error: FindFirstFileExA(c/:/users/{USERNAME}/documents/pico/pico-sdk/src/common/pico_stdlib/include/pico): The filename, directory name, or volume label syntax is incorrect.
[build] 
[build] Build finished with exit code 1`

In VS Code if I do Clean or Clean and Reconfigure I get the same error!

lurch, 2021-01-24:
Seems to be several problems with `ninja`, see https://github.com/raspberrypi/pico-sdk/issues/5 for details on switching to `nmake`.

aallan, 2021-01-24:
I never tested with ninja at all, all the Windows instructions are using nmake. Looks like the CMake plugin may have changed between writing the Windows instructions and last week.

james-rich, 2021-01-24:
Thanks this solved the issue

closed 2021-01-24 by james-rich

cgudrian, 2021-03-13:
I was facing the same problem. I solved it by upgrading from Ninja 1.10.0 to 1.10.1.

jonathangjertsen, 2021-05-18:
Even though nmake exists, I would recommend maintaining the ninja support since it is **much** faster than nmake. On my machine, `nmake hello_adc hello_multicore hello_pio` takes 1 minute 41 seconds from a clean build folder, while the corresponding ninja command takes 20 seconds.

lurch, 2021-05-18:
You're free to try ninja "at your own risk" but it's not something (at this stage) that we'll be officially supporting.
There's a bit more detail about the "problem" in #5

jonathangjertsen, 2021-05-18:
I understand. It works great though! All the targets in the pico-examples repo build fine, and the ones I have tested run fine as well.

lurch, 2021-05-18:
Good to know :smiley: If there are any places where we inadvertently 'break' things for ninja-build, I suspect we _might_ accept a "fix" as long as it has no negative-impact on the cmake/make/nmake build.

kilograham, 2021-05-18:
Yes, Ninja is **so** much faster than _NMake_ it isn't even funny, so it continuing to work is good (really there is no reason it shouldn't as it is usually a pretty stable CMake generator). In terms of support it is just another level of things to go wrong on Windows :-)