raspberrypi/pico-sdk #13: Install C++ ask on Mac OS m1

hpwit, 2021-01-24:
Hello
I cannot execute the commands sudo apt on Mac OSX big sur m1. I cannot do the full install. Any way of doing that ?
Thank you

aallan, 2021-01-24:
Please see the Section 8.1 of "Getting Started with Raspberry Pi Pico" which talks about installing the SDK on macOS.

hpwit, 2021-01-25:
hello thank uyou for you answer
i have this now

/usr/local/bin/arm-none-eabi-gcc-9.2.1   -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk -o CMakeFiles/cmTC_820b5.dir/testCCompiler.c.o -c testCCompiler.c
[cmake]     arm-none-eabi-gcc-9.2.1: error: arm64: No such file or directory
[cmake]     arm-none-eabi-gcc-9.2.1: error: unrecognized command line option '-arch'; did you mean '-march='?

do you know what to do

closed 2021-01-25 by hpwit

aallan, 2021-01-25:
I guess #5 on macOS is related (same ninja issue)

hpwit, 2021-01-25:
> I guess #5 on macOS is related (same ninja issue)

Yes indeed just instead of NMake we need to use Unix Makefiles. Thank you for the fast answer. Now I am waiting for my boards ;)

aallan, 2021-01-25:
Cool. I'll get the PDF documentation updated for the next release.