raspberrypi/pico-sdk #17: Missing CMake Target For Generating Static C Libraries

napperley, 2021-01-25:
With the SDK there is no CMake target for generating static C libraries. For some programming languages like Kotlin it needs the header files, and the dynamic/static library files in order to do [interop with the C libraries](https://kotlinlang.org/docs/curl.html). Rust also has [similar requirements](https://rust-embedded.github.io/book/interoperability/c-with-rust.html).

The maintainers need to decide if the SDK will be **accessible** to programming languages other than C/C++ like Kotlin, and Rust for example.

msink, 2021-01-25:
Well, usually it works just opposite - Kotlin (or Rust, or whatever) creates static library (*.a) file, and that file is linked by SDK to create flashable image file.

JamesH65, 2021-01-25:
> Well, usually it works just opposite - Kotlin (or Rust, or whatever) creates static library (*.a) file, and that file is linked by SDK to create flashable image file.

That is certainly the preferred approach. Just having a static library of the SDK linked in to your "kotlin" is unlikely to work.

napperley, 2021-01-25:
> > Well, usually it works just opposite - Kotlin (or Rust, or whatever) creates static library (*.a) file, and that file is linked by SDK to create flashable image file.
> 
> That is certainly the preferred approach. Just having a static library of the SDK linked in to your "kotlin" is unlikely to work.

Wouldn't the **elf2uf2** tool ([included](https://github.com/raspberrypi/pico-sdk/tree/master/tools/elf2uf2) in the SDK) fix the problem?

kilograham, 2021-01-25:
No; elf2uf2 just translates the mapped segments in the ELF into UF2 sectors, it doesn't modify the binary itself

kilograham, 2021-01-25:
It is hard to help you further when you haven't answered any of the questions that have been asked.

What do you need?

1) Runtime startup
2) C-runtime startup
3) C-runtime libraries
etc.

If you are just taking a native application and hoping to link it for the RP2040/Pico then, once again, far and away the simplest/best thing to do is to create a static library with your stuff, and link it into a binary using the SDK build

Anything else requires more thought, and more information

napperley, 2021-01-25:
> It is hard to help you further when you haven't answered any of the questions that have been asked.
> 
> Asked another way, what do you need?
> 
> 1. Runtime startup
> 2. C-runtime startup
> 3. C-runtime libraries
>    etc.
> 
> If you are just taking a native application and hoping to link it for the RP2040/Pico then, once again, far and away the simplest/best thing to do is to create a static library with your stuff, and link it into a binary using the SDK build
> 
> Anything else requires more thought, and more information

In response to your questions:

1. Standard Kotlin library, Kotlin Native runtime, and KotlinX Cinterop library (essential for C interop especially with C pointers) statically linked into the static library
2. Can you please elaborate on this question
3. All the high level static libraries from the Raspberry Pi Pico SDK

Since static libraries don't have an entry point how is the SDK going to link a static library with a binary? What is providing the binary that knows how to call the static library, and what is the standard structure used by a static library?

kilograham, 2021-01-25:
Ok, so you arent needing to do something low level like Rust, you are basically a C app with some extra libraries (and static compiled Kotlin code)?

To link the static library, i had intended that you create a bootstrap main() method app which calls a method in your static library with calls your main

napperley, 2021-01-25:
> Ok, so you arent needing to do something low level like Rust, you are basically a C app with some extra libraries (and static compiled Kotlin code)?

Correct. Kotlin unlike Rust is a high level, statically typed programming language. With Linux for example Kotlin can be used to do high level Linux development (aka **user space** programs), and Kotlin's C interop is deep enough that many existing tools which support C programs can be used with Kotlin programs (**readelf**, **gdb**, **lldb** etc).

> 
> To link the static library, i had intended that you create a bootstrap main() method app which calls a method in your static library with calls your main

Sounds like some thought needs to be put into the SDK with how the embedded software life cycle works, and how to architect embedded programs. Your proposed structure is ok, however it would be better to call the method **onApplicationStart** since that method handles the event when the application starts up. The name **main** is a bit confusing since it implies a entry point in a program, when in fact a static library is being used as part of the bootstrapping process.

kilograham, 2021-01-26:
Yup, I know what Kotlin is (ardent IntelliJ user), however I haven't looked at Kotlin native, so was asking.

We provide scaffolding that does c-runtime initialization and calls main(), because that seemed friendly to C users. I assume your Kotlin runtime starts at main() too. 

You can call your method whatever you like... you said that your static library had no entry point, so I said make one and call it from the main method. 

We have put a lot of thought into the SDK, and we won't be feeling the need to change the name to onApplicationStart.

msink, 2021-01-26:
Currently Kotlin/Native does not support baremetall applications in default distribution, but two years ago it experimented with Zephyr RTOS on STM32F4 board (essentially same as RPi Pico)
LED blinking example: https://github.com/JetBrains/kotlin-native/blob/master/samples/zephyr/src/main.kt

kilograham, 2021-01-26:
That's cool; it shouldn't be too hard to get going, and we'd love to help more, but we're working on _lots_ of things right now!

The SDK works the way it does with good reasons; it isn't the same for embedded as it is for your desktop compile source, link against a library and run. However we designed the CMake build to hide *lots* of complexity from you and build nicely tuned binaries off the bat, and give the user the power to cleanly have finer control if they want. A c user is very familiar with a main() method that runs their  code.

You can read https://datasheets.raspberrypi.org/pico/sdk/pico_c_sdk.pdf for some intro on this.

napperley, 2021-01-26:
> We provide scaffolding that does c-runtime initialization and calls main(), because that seemed friendly to C users. I assume your Kotlin runtime starts at main() too.

Yes, although the type signature(s) are different from the C version:

**Plain Version**

```kotlin
fun main() {
    println("Program with no arguments")
}
```

**Program Arguments Version**

```kotlin
fun main(args: Array<String>) {
    println("Program with arguments: $args")
}
```

msink, 2021-01-26:
See Kotlin/Native runtime: https://github.com/JetBrains/kotlin-native/blob/1f0dd817421aba7a044d623094fe584d609fb68b/runtime/src/launcher/cpp/launcher.cpp#L66

kilograham, 2021-04-07:
inactive

closed 2021-04-07 by kilograham