raspberrypi/pico-sdk #22: SVD FIle -missing VTOR table under CPU

conkerkh, 2021-01-26:
From SVD file specification under CPU section, vtorPresent is missing. According to the datasheet of RP2040 VTOR relocation using VTOR register is supported. Please add vtorPresent field set to 1 under CPU section.

[https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_cpu.html](url)

Wren6991, 2021-02-03:
Note this is probably the reason it wasn't added:

> vtorPresent | Indicate whether the Vector Table Offset Register (VTOR) is implemented in Cortex-M0+ based devices. This tag is either set to true or false, 1 or 0. **If not specified, then VTOR is assumed to be present.**

Raised a PR to add this anyway as there's no need to be ambiguous

Wren6991, 2021-02-03:
Now merged

closed 2021-02-03 by Wren6991