raspberrypi/pico-sdk #23: pioasm/pioasm.exe is missing and no known rule to make it

vishnumaiea, 2021-01-26:
I was trying to build the examples from VS Code and encountered the following error.

```
[main] Building folder: pico-examples 
[main] Configuring folder: pico-examples 
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -Hc:/Pico/pico-examples -Bc:/Pico/pico-examples/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] Using PICO_SDK_PATH from environment ('C:\Pico\pico-sdk')
[cmake] Pico SDK is located at C:/Pico/pico-sdk
[cmake] Defaulting PICO_PLATFORM to rp2040 since not specified.
[cmake] Defaulting PICO platform compiler to pico_arm_gcc since not specified.
[cmake] PICO compiler is pico_arm_gcc
[cmake] PICO_GCC_TRIPLE defaulted to arm-none-eabi
[cmake] -- The C compiler identification is GNU 9.3.1
[cmake] -- The CXX compiler identification is GNU 9.3.1
[cmake] -- The ASM compiler identification is GNU
[cmake] -- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/9 2020-q2-update/bin/arm-none-eabi-gcc.exe
[cmake] Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
[cmake] Defaulting PICO target board to pico since not specified.
[cmake] Using board configuration from C:/Pico/pico-sdk/src/boards/include/boards/pico.h
[cmake] -- Found Python3: C:/Users/Vishnu Mohanan/AppData/Local/Programs/Python/Python39/python.exe (found version "3.9.1") found components: Interpreter 
[cmake] TinyUSB available at C:/Pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
[cmake] Compiling TinyUSB with CFG_TUSB_DEBUG=1
[cmake] -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
[cmake] ELF2UF2 will need to be built
[cmake] PIOASM will need to be built
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: C:/Pico/pico-examples/build
[build] Starting build
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.EXE" --build c:/Pico/pico-examples/build --config Debug --target all -- -j 18
[build] ninja: error: 'pioasm/pioasm.exe', needed by 'adc/dma_capture/CMakeFiles/adc_dma_capture_resistor_dac_pio_h', missing and no known rule to make it
[build] Build finished with exit code 1
```
All dependencies are installed with correct path variables. The machine is Windows 10.

lurch, 2021-01-26:
`... -G Ninja`

Please see the recently-updated Getting Started document for details on switching from Ninja to Makefiles.

aallan, 2021-01-26:
See #5 and the associated changed in the [Getting Started](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) book made today.

closed 2021-01-26 by aallan