raspberrypi/pico-sdk #24: NMake Fails to Build the Project

vishnumaiea, 2021-01-27:
Well, we have a new problem and I couldn't find anybody referenced this in any reported issues.

Setup : VS Code, Windows 10 x64.

Tried building the examples from Developer Command Prompt for VC 2019. NMake makefile generation was success, but building from them failed with the following output.

```
C:\Pico\pico-examples\build>nmake

Microsoft (R) Program Maintenance Utility Version 14.28.29336.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Scanning dependencies of target bs2_default
[  0%] Building ASM object pico_sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/boot2_w25q080.S.obj
[  0%] Linking ASM executable bs2_default.elf
[  0%] Built target bs2_default
Scanning dependencies of target bs2_default_bin
[  0%] Generating bs2_default.bin
[  0%] Built target bs2_default_bin
Scanning dependencies of target bs2_default_padded_checksummed_asm
[  0%] Generating bs2_default_padded_checksummed.S
[  0%] Built target bs2_default_padded_checksummed_asm
Scanning dependencies of target ELF2UF2Build
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] No download step for 'ELF2UF2Build'
[  0%] No update step for 'ELF2UF2Build'
[  0%] No patch step for 'ELF2UF2Build'
[  0%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is MSVC 19.28.29336.0
-- The CXX compiler identification is MSVC 19.28.29336.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.28.29333/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.28.29333/bin/Hostx86/x86/cl.exe - broken
CMake Error at C:/Program Files/CMake/share/cmake-3.19/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.28.29333/bin/Hostx86/x86/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Pico/pico-examples/build/elf2uf2/CMakeFiles/CMakeTmp

    Run Build Command(s):nmake /nologo cmTC_171d1\fast &&       "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29333\bin\HostX86\x86\nmake.exe"  -f CMakeFiles\cmTC_171d1.dir\build.make /nologo -L                  CMakeFiles\cmTC_171d1.dir\build
    Building C object CMakeFiles/cmTC_171d1.dir/testCCompiler.c.obj
        C:\PROGRA~2\MICROS~2\2019\BUILDT~1\VC\Tools\MSVC\1428~1.293\bin\Hostx86\x86\cl.exe @C:\Users\VISHNU~1\AppData\Local\Temp\nm9649.tmp
    testCCompiler.c
    Linking C executable cmTC_171d1.exe
        "C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_171d1.dir --rc=rc --mt=CMAKE_MT-NOTFOUND --manifests -- C:\PROGRA~2\MICROS~2\2019\BUILDT~1\VC\Tools\MSVC\1428~1.293\bin\Hostx86\x86\link.exe /nologo @CMakeFiles\cmTC_171d1.dir\objects1.rsp @C:\Users\VISHNU~1\AppData\Local\Temp\nm9698.tmp
    RC Pass 1: command "rc /fo CMakeFiles\cmTC_171d1.dir/manifest.res CMakeFiles\cmTC_171d1.dir/manifest.rc" failed (exit code 0) with the following output:
    The system cannot find the file specifiedNMAKE : fatal error U1077: '"C:\Program Files\CMake\bin\cmake.exe"' : return code '0xffffffff'
    Stop.
    NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29333\bin\HostX86\x86\nmake.exe"' : return code '0x2'
    Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Pico/pico-examples/build/elf2uf2/CMakeFiles/CMakeOutput.log".
See also "C:/Pico/pico-examples/build/elf2uf2/CMakeFiles/CMakeError.log".
NMAKE : fatal error U1077: 'echo' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29333\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29333\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
```

The same error appears when trying to build with VS Code also. Looks like some "not found" error. What could be wrong ?

Reading from the error log, the culprit could be,

`LINK : fatal error LNK1104: cannot open file 'kernel32.lib'`

[CMakeError.log](https://github.com/raspberrypi/pico-sdk/files/5877531/CMakeError.log)
[CMakeOutput.log](https://github.com/raspberrypi/pico-sdk/files/5877534/CMakeOutput.log)

vishnumaiea, 2021-01-27:
With a little investigation, I found this suggestion : https://stackoverflow.com/questions/43422105/lnk1104-cannot-open-file-kernel32-lib-when-windows-sdk-version-is-set-to-15 which says,

```
The Windows 10 Creators Update SDK made some significant changes to what is installed as part of the installation to minimize on-disk footprint. This is a Known Issue that results from that refactoring work where the "desktop" components of the Creators Update SDK do not get installed by default. The previous new project logic always selected the latest Windows 10 SDK, assuming the desktop portions were installed as part of that installation scenario. We are working on a fix for a future update.

The workarounds available are as follows:

Select a version of the Windows 10 SDK that's fully installed on the system (i.e. 10.0.14393.0 or earlier) in the Project Properties

Install the 'Windows 10 SDK (10.0.15063.0) for Desktop' component via the Visual Studio Installer. win10sdk-15063desktop.png

Daniel Griffing,

Visual C++ Libraries
```

I had skipped installing the **Windows 10 SDK** from the VS build tool installation step. I am now installing the Windows SDK to see if it solves the problem.

vishnumaiea, 2021-01-27:
Yes, installing the previously skipped Windows 10 SDK solved the problem.

[image]

closed 2021-01-27 by vishnumaiea

vishnumaiea, 2021-01-27:
Looks like it's now Arduino's job to make everything easy; from installing all dependencies, to building and uploading a project with a single press of a button.

fattymatty195, 2021-02-19:
To get mine past this error, I had to run the nmake command from visual studio developer command prompt. (I was using visual studio 2019 professional)

lurch, 2021-02-19:
@vishnumaiea We forgot to come back and update this issue, but we added a note to the latest version of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf explaining that the full "Windows 10 SDK" package needs to be installed. Thanks for your feedback.

makingsoundmachines, 2022-01-19:
> To get mine past this error, I had to run the nmake command from visual studio developer command prompt. (I was using visual studio 2019 professional)

This helped me as well.

@vishnumaiea 's find suggested installing 'Windows 10 SDK (**10.0.15063.0**) for Desktop'. The earliest version I could install from my Visual Build Tools installer was 10.016299.0