raspberrypi/pico-sdk #25: Unable to download the SDK

ibrahimqazi, 2021-01-27:
Hello dear maintainers , I am unable to download the SDK. Please help.
This command works:
`git clone -b master https://github.com/raspberrypi/pico-sdk.git`
but when I enter `git submodule update --init`
It ends with no success.
```
Failed to clone 'lib/tinyusb'. Retry scheduled
Cloning into 'D:/Education/Tutorials/raspberry pi pico/sdk_examples/pico-sdk/lib/tinyusb'...
```

lurch, 2021-01-27:
I just tested it (although admittedly on Linux rather than on Windows) and it works for me. Perhaps it was a temporary failure? Can you try again?

ibrahimqazi, 2021-01-28:
> I just tested it (although admittedly on Linux rather than on Windows) and it works for me. Perhaps it was a temporary failure? Can you try again?

Thanks for your reply. I tried on Linux system as well but still no luck.
`Submodule 'tinyusb' (https://github.com/raspberrypi/tinyusb.git) registered for path 'lib/tinyusb'
Cloning into '/media/ibrrahim/New Volume/Education/Tutorials/raspberry pi pico/sdk_examples/pico-sdk/lib/tinyusb'...
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/raspberrypi/tinyusb.git' into submodule path '/media/ibrrahim/New Volume/Education/Tutorials/raspberry pi pico/sdk_examples/pico-sdk/lib/tinyusb' failed
Failed to clone 'lib/tinyusb'. Retry scheduled
Cloning into '/media/ibrrahim/New Volume/Education/Tutorials/raspberry pi pico/sdk_examples/pico-sdk/lib/tinyusb'...
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/raspberrypi/tinyusb.git' into submodule path '/media/ibrrahim/New Volume/Education/Tutorials/raspberry pi pico/sdk_examples/pico-sdk/lib/tinyusb' failed
Failed to clone 'lib/tinyusb' a second time, aborting`

lurch, 2021-01-28:
That's really bizarre. Just out of curiosity, are you able to "manually" `git clone https://github.com/raspberrypi/tinyusb.git` into a separate directory?

> error: RPC failed; curl 18 transfer closed with outstanding read data remaining

Are you using a proxy or something to access the internet?

ibrahimqazi, 2021-01-29:
> 
> 
> That's really bizarre. Just out of curiosity, are you able to "manually" `git clone https://github.com/raspberrypi/tinyusb.git` into a separate directory?
> (EDIT: If that doesn't work, see if `git clone https://github.com/raspberrypi/tinyusb` works?)
> 
> > error: RPC failed; curl 18 transfer closed with outstanding read data remaining
> 
> Are you using a proxy or something to access the internet?

Finally after several attempts I downloaded it using GitHub Desktop. I tried with and without proxy.

closed 2021-01-29 by ibrahimqazi