raspberrypi/pico-sdk #27: gettimeofday() does not work

Hermann-SW, 2021-01-27:
struct timeval seconds are always 0, regardless how much time passed (also for sleep_ms(3333)):
```
tud_cdc_connected()                                    
0 / 536877456                                          
0 / 3758157056                                         
0 / 536873932

```

Code:
```
#include <stdio.h>
#include "pico/stdlib.h"
#include <tusb.h>
#include <sys/time.h>

int main(void)
{
  stdio_init_all();
  while (!tud_cdc_connected()) { sleep_ms(100);  }
  printf("tud_cdc_connected()\n");

  struct timeval tv0,tv1,tv2;
  gettimeofday(&tv0, NULL);
  sleep_ms(1);
  gettimeofday(&tv1, NULL);
  sleep_ms(2);
  gettimeofday(&tv2, NULL);

  printf("%ld / %u\n",tv0.tv_sec,tv0.tv_usec);
  printf("%ld / %u\n",tv1.tv_sec,tv1.tv_usec);
  printf("%ld / %u\n",tv2.tv_sec,tv2.tv_usec);

  return 0;
}

```

kilograham, 2021-01-28:
Note that this is actually a newlib issue.

It does return -1 for an error, and sets errno to ENOSYS (i.e. not implemented)

kilograham, 2021-04-28:
won't fix

kilograham, 2021-04-28:
this is trivial to fix, so may as well

kilograham, 2021-05-24:
moving to 1.3.0 with other newlib stuff

kilograham, 2023-01-23:
fixed by #1186

kilograham, 2023-01-24:
merged into develop

closed 2023-01-24 by kilograham