raspberrypi/pico-sdk #28: pico_time/time.c using GPIO pins

sde1000, 2021-01-27:
I've been having a problem with calls to functions in time.c (eg. `sleep_ms()`) having side effects on GPIO pins.

I'm pretty sure this is related to the DEBUG_PINS macros; changing `#define PICO_DEBUG_PIN_BASE 19u` in src/rp2_common/hardware_gpio/include/hardware/gpio.h to something else changes the GPIO pins that are affected. (My project is using GPIO 21 as an output, which is how I discovered this.)

I don't immediately see how to turn this off!

sde1000, 2021-01-27:
[time.c](https://github.com/raspberrypi/pico-sdk/blob/master/src/common/pico_time/time.c) lines 16–17 is currently:
```
CU_REGISTER_DEBUG_PINS(core)
CU_SELECT_DEBUG_PINS(core)
```
This appears to be enough for `DEBUG_PINS_ENABLED(core)` to be true and for the DEBUG_PINS_foo(core, v) macros to call `gpio_set_mask()` etc. as appropriate, whether or not `gpio_debug_pins_init()` has been called at any point.

I think the correct fix is to comment out `CU_SELECT_DEBUG_PINS(core)` in time.c

kilograham, 2021-01-27:
Doh! Yes that should not be enabled; will fix (and grep for others) - that probably snuck back in while testing something!

PeterOGB, 2021-01-27:
Does this mean that gpio_debug_pins_init()  also doen't work as intended inso far as it is not the only way to enable the debug ppins ?

kilograham, 2021-01-27:
Nope, gpio_debug_pins_init() sets up the pins for output. The macros turn on/off code that writes values to the pins for debugging (mostly for use with logic analyzers when debugging higher frequency stuff)

the macros should not be enabled in any library code.

PeterOGB, 2021-01-27:
Ok, but you did write "for the debug pins to do anything, you have to call gpio_debug_pins_init()" which sounded like it enabled them as well as setting them to output  and l can't see where gpio_debug_pins_init() is covered in the SDK documentation.

kilograham, 2021-01-31:
merged into develop

closed 2021-01-31 by kilograham

aallan, 2021-02-02:
This should now be fixed in MicroPython mainline, see micropython/micropython#6826. Nightly UF2 builds of MicroPython are available from https://micropython.org/download/rp2-pico/.