raspberrypi/pico-sdk #46: Incorrect attribute for structure member of SysTick registers

djrose80, 2021-01-29:
Looking at the file src/rp2040/hardware_structs/include/hardware/structs/systick.h

The member cvr is of type io_ro_32 but it is a read/write register (as detailed in section 2.4.8 of rp2040-datasheet.pdf, and general Cortex M0 literature).

lurch, 2021-01-29:
@kilograham It's correctly marked as RW in https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_regs/include/hardware/regs/m0plus.h#L106 so I guess this is just a typo in https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_structs/include/hardware/structs/systick.h#L16 ?

lurch, 2021-03-02:
Seems that this was fixed by #91 so I guess this can be closed?

closed 2021-03-02 by aallan