raspberrypi/pico-sdk #50: enforce that gcc toolchain only used thumb code

cleverca22, 2021-01-29:
on my [distro](https://nixos.org) of choice, the gcc currently lacks a thumb version of `crtbegin.o`, so it silently inserted an arm32 version into all of my pico builds, resulting in nothing working
fixing that problem is outside of the scope of pico-sdk

but, detecting it, and warning/failing is probably within the scope of pico-sdk, and is very simple to do:
```
[nix-shell:~/apps/rpi/pico/pico-examples]$ readelf --arch-specific build/hello_world/usb/hello_usb.elf
Attribute Section: aeabi
File Attributes
  Tag_CPU_name: "ARM v6K"
  Tag_CPU_arch: v6K
  Tag_CPU_arch_profile: Microcontroller
  Tag_ARM_ISA_use: Yes
  Tag_THUMB_ISA_use: Thumb-1
  Tag_ABI_PCS_wchar_t: 4
  Tag_ABI_FP_denormal: Needed
  Tag_ABI_FP_exceptions: Needed
  Tag_ABI_FP_number_model: IEEE 754
  Tag_ABI_align_needed: 8-byte
  Tag_ABI_enum_size: small

[nix-shell:~/apps/rpi/pico/pico-examples]$ readelf --arch-specific build-ubuntu/hello_world/usb/hello_usb.elf
Attribute Section: aeabi
File Attributes
  Tag_CPU_name: "6S-M"
  Tag_CPU_arch: v6S-M
  Tag_CPU_arch_profile: Microcontroller
  Tag_THUMB_ISA_use: Thumb-1
  Tag_ABI_PCS_wchar_t: 4
  Tag_ABI_FP_denormal: Needed
  Tag_ABI_FP_exceptions: Needed
  Tag_ABI_FP_number_model: IEEE 754
  Tag_ABI_align_needed: 8-byte
  Tag_ABI_enum_size: small
```
when gcc did the wrong thing, it added a `Tag_ARM_ISA_use: Yes` into the arch specific flags on the resulting elf binary
it should be pretty simple to just check those flags after linking, and then fail if gcc somehow put arm32 code into the final elf

cleverca22, 2021-01-30:
update:
ive learned that the critical difference is if gcc was built with multilib
an ubuntu docker container:
```
root@bb6258570479:~# arm-none-eabi-gcc -print-multi-lib                                                                                                                                                             
.;                                                                                                                                                                                                                  
arm/v5te/softfp;@marm@march=armv5te+fp@mfloat-abi=softfp                                                                                                                                                            
arm/v5te/hard;@marm@march=armv5te+fp@mfloat-abi=hard                                                                                                                                                                
thumb/nofp;@mthumb@mfloat-abi=soft                                                                                                                                                                                  
thumb/v7/nofp;@mthumb@march=armv7@mfloat-abi=soft                                                                                                                                                                   
thumb/v7+fp/softfp;@mthumb@march=armv7+fp@mfloat-abi=softfp                                                                                                                                                         
thumb/v7+fp/hard;@mthumb@march=armv7+fp@mfloat-abi=hard                                                                                                                                                             
thumb/v7-r+fp.sp/softfp;@mthumb@march=armv7-r+fp.sp@mfloat-abi=softfp                                                                                                                                               
thumb/v7-r+fp.sp/hard;@mthumb@march=armv7-r+fp.sp@mfloat-abi=hard                                                                                                                                                   
thumb/v6-m/nofp;@mthumb@march=armv6s-m@mfloat-abi=soft                                                                                                                                                              
thumb/v7-m/nofp;@mthumb@march=armv7-m@mfloat-abi=soft                                                                                                                                                               
thumb/v7e-m/nofp;@mthumb@march=armv7e-m@mfloat-abi=soft                                                                                                                                                             
thumb/v7e-m+fp/softfp;@mthumb@march=armv7e-m+fp@mfloat-abi=softfp                                                                                                                                                   
thumb/v7e-m+fp/hard;@mthumb@march=armv7e-m+fp@mfloat-abi=hard                                                                                                                                                       
thumb/v7e-m+dp/softfp;@mthumb@march=armv7e-m+fp.dp@mfloat-abi=softfp                                                                                                                                                
thumb/v7e-m+dp/hard;@mthumb@march=armv7e-m+fp.dp@mfloat-abi=hard                                                                                                                                                    
thumb/v8-m.base/nofp;@mthumb@march=armv8-m.base@mfloat-abi=soft                                                                                                                                                     
thumb/v8-m.main/nofp;@mthumb@march=armv8-m.main@mfloat-abi=soft                                                                                                                                                     
thumb/v8-m.main+fp/softfp;@mthumb@march=armv8-m.main+fp@mfloat-abi=softfp                                                                                                                                           
thumb/v8-m.main+fp/hard;@mthumb@march=armv8-m.main+fp@mfloat-abi=hard                                                                                                                                               
thumb/v8-m.main+dp/softfp;@mthumb@march=armv8-m.main+fp.dp@mfloat-abi=softfp                                                                                                                                        
thumb/v8-m.main+dp/hard;@mthumb@march=armv8-m.main+fp.dp@mfloat-abi=hard                                                                                                                                            
```
the broken nixos gcc:
```
[clever@amd-nixos:~/apps/nixpkgs]$ before/bin/arm-none-eabi-gcc -print-multi-lib
.;
```
after rebuilding nixos's gcc with `--with-multilib` added to the configure flags:
```
[clever@amd-nixos:~/apps/nixpkgs-hasura]$ after/bin/arm-none-eabi-gcc -print-multi-lib
.;
thumb;@mthumb
arm/autofp/v5te/fpu;@marm@mfpu=auto@march=armv5te+fp@mfloat-abi=hard
thumb/autofp/v7/fpu;@mthumb@mfpu=auto@march=armv7+fp@mfloat-abi=hard
```
the arm32 code from `crtbegin.o` at `deregister_tm_clones` has been corrected to thumb, but `Tag_ARM_ISA_use: Yes` is still present

will debug further

cleverca22, 2021-01-30:
update2:

the gcc must be built with `--enable-multilib --with-multilib-list=rmprofile`, and the newlib must be built in either multilib (or at least thumb mode), and the `arm-none-eabi/lib/thumb/v6-m/nofp` of the newlib must be in the linker's search path

`readelf --arch-specific` no longer reports `Tag_ARM_ISA_use`, and the resulting binaries now correctly function on my pico

the `Tag_ARM_ISA_use` flag definitely looks like a strong sign that your toolchain is broken and the binary wont work on an rp2040

kilograham, 2021-04-28:
closing as inactive; if you want to update elf2uf2 open a PR

closed 2021-04-28 by kilograham

lurch, 2021-04-28:
This links to #70 (where there are more comments)