raspberrypi/pico-sdk #51: Values of the _SELECTED registers in the clock controllers not explained

Tiwalun, 2021-01-29:
I was not able to find any documentation for the values of the _SELECTED registers in the clock controllers.

In the SVD file, they are simply described as 32-bit registers, but not specified in more detail. The other clock controller registers,
like the _CTRL registers, have `fields` with enumerated values. It would be very nice to have this for the _SELECTED registers as well.

The datasheet does also not offer more help, it says that the values are one-hot encoded, but it doesn't say what the bits mean.

Is this documented somewhere?

Wren6991, 2021-02-03:
There is some description in the datasheet in `2.15.3.2 Multiplexers`:
[image]

The switch of the glitchless mux is not instantaneous (if it was, it would cause glitches) and the SELECTED register lets you poll until your selection has actually propagated through the clock hardware.

Seems like there is a discoverability issue here so I'll make sure there is some more info in the headers and in the datasheet register listings.

Wren6991, 2021-02-03:
Oh sorry I missed that this was a specific question about the values read from the registers -- it's a one-hot decode of the clocks enumerated in the `SRC` field of the `CTRL` register. I'll make sure that is called out more explicitly.

[image]

Tiwalun, 2021-02-03:
I think it's a bit confusing because it's split into two fields, so it's not immediately clear how the clocks are mapped.

I would assume that for `clk_sys` the meaning is like this:
- 1 = `clk_ref`
- 2 = `clksrc_pll_sys`
- 3 = `clksrc_pll_usb`
- ...

Is this correct?

lurch, 2021-02-03:
_If_ I'm understanding what's written above correctly, CLK_SYS_SELECTED would decode as:
* 1 = clk_ref
* 2 = clksrc_slk_sys_aux

Sounds like changing the AUXSRC glitches, which means it switches immediately, and therefore it has no need for a SELECTED status register?

Wren6991, 2021-02-03:
> Sounds like changing the AUXSRC glitches, which means it switches immediately, and therefore it has no need for a SELECTED status register?

Yep, this is only for the glitchless mux, which is not present on most of the clock slices. I think what is really missing here docs wise is a block diagram in `2.15.3.2 Multiplexers` to make it apparent there are two layers of muxing on some of the slices, and explain why it is needed (it's just there for clocks which need to be switched on the fly, as oppose to stopping, switching and restarting)

closed 2021-05-31 by kilograham