raspberrypi/pico-sdk #55: Ethernet?

BBBSnowball, 2021-01-31:
Is there any support for Ethernet, yet?

The PIOs should be capable of talking to 100 Mbit PHYs. I have a working prototype for the receive side with RMII and a LAN8720 PHY: https://github.com/BBBSnowball/pico-examples/tree/master/pio/rmii

Is this something we would want to have in pico-sdk or pico-extras? Are there any existing efforts that I have overlooked? I can hardly believe that I'm the only one working on this ;)

aallan, 2021-02-01:
There is a USB Ethernet example in the rasberrypu/pico-playground repo.

closed 2021-02-01 by aallan

BBBSnowball, 2021-02-01:
This is only speaking ethernet over USB, right? I want to add actual Ethernet, like this project for ESP32: https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

The ESP32 has a MAC peripheral so attaching a PHY is quite easy and supported by their SDK. I guess, my code should go into a fork of extras or playground.