raspberrypi/pico-sdk #72: FPGA related code still needed?

SteveGotthardt, 2021-02-02:
I was looking at blink..dis and found:
10000460 <running_on_fpga>:
10000460:	4b02      	ldr	r3, [pc, #8]	; (1000046c <running_on_fpga+0xc>)
10000462:	6818      	ldr	r0, [r3, #0]
10000464:	0780      	lsls	r0, r0, #30
10000466:	0fc0      	lsrs	r0, r0, #31
10000468:	4770      	bx	lr
1000046a:	46c0      	nop			; (mov r8, r8)
1000046c:	4006c000 	.word	0x4006c000

It is defined here:
https://github.com/raspberrypi/pico-sdk/blob/2d5789eca89658a7f0a01e2d6010c0f254605d72/src/rp2_common/pico_platform/platform.c#L12

and is called here:
https://github.com/raspberrypi/pico-sdk/blob/2d5789eca89658a7f0a01e2d6010c0f254605d72/src/rp2_common/hardware_clocks/clocks.c#L122

What to do?
> add define like RUNNING_ON_FPGA
> remove FPGA referring code?
> leave in for debug and remove for release?
> or?

cleverca22, 2021-02-02:
i'm thinking it could be something like this:
```
#if NOT_RUNNING_ON_FPGA
static inline bool running_on_fpga() { return false; }
#else
bool running_on_fpga();
#endif
```
the compiler should then do some static analysis, and realize that half of the if statement can never run, and then delete the dead code

Wren6991, 2021-02-02:
We can add a define to stub out the check, looks like that would save around 24 bytes if you include the call and test. I'd prefer if it stays in by default, so that when we get a repro binary for a potential hardware issue, we can immediately spin it up on the FPGA and capture logic traces, without having to perturb the binary layout.

kilograham, 2021-02-02:
^ that is why it is the way it is... 
`#if !PICO_NO_FPGA_CHECK`

is our idiomatic way of doing this

Wren6991, 2021-02-04:
You can now add `PICO_NO_FPGA_CHECK=1` to your compile definitions if you're building against the SDK `develop` branch

Wren6991, 2021-02-05:
Comment from OP on the PR that was merged:

> Good solution. Thanks

Taking that as permission to close :)

closed 2021-02-05 by Wren6991