raspberrypi/pico-sdk #76: USB GP15 errata fix should restore pin mux state

tannewt, 2021-02-02:
The USB errata code [overrides the pin mux of GP15](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c#L81) in order to reset the internal USB peripheral. After it's done with the pin, it doesn't restore the mux setting. It should restore the mux setting back to it's previous state so that any code that was using it, continues to work.

liamfraser, 2021-02-02:
I agree. There is even a TODO in the code to say to do this. Is CircuitPython definitely built with this errata fix included? The SDK does not include it by default.

tannewt, 2021-02-03:
Yes, I believe we enable it here through tinyusb: https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/Makefile#L111

Wren6991, 2021-02-05:
Fix is now merged into the `develop` branch, @tannewt could you please test it out?

tannewt, 2021-02-05:
I haven't reproduced it myself but will update CP to use the develop branch in the next beta. That'll get a bunch of folks using it.

dhalbert, 2021-02-08:
We incorporated this fix into CircuitPython `main` and three users who had trouble have reported that it fixes their GP14 problems.
See https://github.com/adafruit/circuitpython/issues/4107.

Wren6991, 2021-02-09:
Great! Thank you for the good news. This will appear in the next tagged SDK release too.

closed 2021-02-09 by Wren6991