raspberrypi/pico-sdk #85: Usb assertion "hw_data_offset(next_buffer_ptr) <= USB_DPRAM_MAX"

ndinsmore, 2021-02-04:
This is a duplicate of the issue I created at please see the issue there for sull documentation:
hathach/tinyusb/issues/628

In the issue I created in the tinyusb repo I mention that I am using the new code coming directly from that repo, but the code at that is at issue is in the SDK tiny USB repo as well.

lurch, 2021-02-05:
Sounds like something @liamfraser needs to look at?

ndinsmore, 2021-02-10:
This issue is fixed by hathach/tinyusb#642 @liamfraser it is one real line of code so it should be easy for you to take a look at.

liamfraser, 2021-02-10:
Looks like a plausible fix to me. I think the audio example may be the first one that repeatedly opens and closes an ep which is why we didn't see this?

liamfraser, 2021-02-10:
(I don't own the upstream tinyusb repo so I can't merge the PR)

hathach, 2021-02-10:
@liamfraser please accept my contributor invitation, and feel free to review, approval and merge any PRs that make sense to you. I still haven't got into the usb specs of rp2040 yet. Worse I am on vacation just now :wink:

liamfraser, 2021-02-12:
@ndinsmore I have merged this into tinyusb upstream

liamfraser, 2021-02-12:
Will keep this open until either a) our sdk tracks upstream or b) we pick this commit into our fork. We need to review what changes have been made in tinyusb upstream before switching the SDK to it.

kilograham, 2021-04-08:
closing in favor of #161

kilograham, 2021-04-08:
changed my mind, will get closed with #321

kilograham, 2021-05-31:
actually closed by #462

kilograham, 2021-06-04:
released

closed 2021-06-04 by kilograham