raspberrypi/pico-sdk #87: boot stage2: double reset to uf2

hathach, 2021-02-05:
Hi hi, thank you for the pico. 

It is great to have uf2 built into the bootrom, as maintainer of similar [tinyuf2](https://github.com/adafruit/) (cross-platform uf2 bootloader), I kind of miss the double tap on reset button to enter uf2 feature. Which set a sram location to a magic number and wait for 0.5 second. If another reset occurs (double reset) during wait time then mcu will enter uf2 mode as following code does:

https://github.com/adafruit/tinyuf2/blob/master/src/main.c#L84 

- **pros**: it  requires only 1 button (RST) for small form factor PCB and in general easier to operate than press reset and hold BOOTSEL.
- **cons**: it does introduce additional 0.5s to boot process.

I wonder if RPI would be interested in adding this feature, my approach would be modify boot stage2 for double reset detection then use watchdog reset to jump to UF2. https://github.com/raspberrypi/pico-sdk/tree/master/src/rp2_common/boot_stage2

Thanks

Wren6991, 2021-02-05:
Hi,

We have an example of a library for doing this by injecting before main, here:

https://github.com/raspberrypi/pico-examples/tree/master/system/double_tap_usb_boot

An example of using it is here:

https://github.com/raspberrypi/pico-examples/tree/master/system/hello_double_tap

The second stage is responsible for setting up flash execution -- anything not directly related to that does not need to be in the second stage, and can be registered to run before `main()` with a `constructor` attribute (or by explicitly allocating a function pointer in `.init_array`)

hathach, 2021-02-05:
Oh that is great to hear, sorry, I didn't notice it is already available for usage as library. It is probably better to include it as default, so that user can quickly switch between examples. Though that is totally your call to apply to sdk examples. I will notify other folks at adafruit to include this @tannewt . Thanks for the tips.

Wren6991, 2021-02-05:
We can look at moving it to the SDK and adding a build flag. I'll talk to @kilograham when he is back from holiday

hathach, 2021-02-05:
> We can look at moving it to the SDK and adding a build flag. I'll talk to @kilograham when he is back from holiday

That sounds like a great plan :+1:

hathach, 2021-02-09:
I reopen the issue as reminder for your decision on the SDK discussion. Feel free to close it anytime you make the decision. Thanks

Wren6991, 2021-02-10:
I did ask him last night but will ping here -- @kilograham are you happy to move that library into the SDK? Seems like it would mostly be a positive change, only major problem I see is that you can inadvertently wipe your flash or not include the lib, and you might prefer to still have a BOOTSEL button for those cases.

kilograham, 2021-02-10:
i'm fine with adding the library... name needs some thought; double tap of what? perhaps it is obvious idk

Wren6991, 2021-02-11:
Let's start the bidding at `pico_bootsel_on_double_reset`?

kilograham, 2021-02-11:
not sure if you are making fun of my like of long names, but i'm fine with that (or `pico_bootsel_via_double_reset`)

hathach, 2021-02-12:
Superb news. Thanks for adding this, I am happy with any names 👍 👍

hathach, 2021-02-18:
closed thanks to #137

commit 4c1fac58 (2021-02-18): Add pico_bootsel_via_double_reset library, fix #87 (#137)

* Add pico_bootsel_via_double_reset library, fix #87

* pico_bootsel_via_double_reset: remove unnecessary library guard, fix subdirectory ordering

* Fix C type used as PICO_CONFIG type
closed 2021-02-18 by hathach