raspberrypi/pico-sdk #89: Technical issues in Doxygen configuration

cesarvandevelde, 2021-02-05:
I encountered a couple of smaller Doxygen issues while attempting to build the SDK documentation locally. The proposed fixes are small, but took some fiddling to get right. Everything was tested in the following environment:

* OSX Catalina
* Doxygen v1.9.1
* Cmake v3.19.2
* Pico SDK v1.0.1

# Stray tag in DoxygenLayout.xml

There's a stray `<tab>` tag in `DoxygenLayout.xml`, probably copy-pasted from the `<navindex>` section. The documentation builds fine, but Doxygen issues a warning.

```diff
--- DoxygenLayout.xml	2021-02-03 23:20:40.000000000 +0100
+++ DoxygenLayout.xml	2021-02-04 23:11:51.000000000 +0100
@@ -146,8 +146,7 @@
   </namespace>

   <!-- Layout definition for a file page -->
-  <file>    <tab type="modules" visible="yes" title="Libraries" intro="Here is a list of all the libraries supported in the Raspberry Pi Pico SDK"/>
-
+  <file>
     <briefdescription visible="yes"/>
     <includes visible="$SHOW_INCLUDE_FILES"/>
     <includegraph visible="$INCLUDE_GRAPH"/>
```

# Missing briefs

The current configuration does not pick up briefs for most groups. The resulting API overview page looks like this:

[image]

This can be fixed by adding `JAVADOC_AUTOBRIEF = YES` to `Doxyfile.in`. Alternatively, the `\brief` annotation could be added to the source code documentation.

# Configured project number is overwritten

The variable `PROJECT_NUMBER` is defined twice in `Doxyfile.in`. The resulting version number is hard-coded to 1.0.

```
PROJECT_NUMBER         = @VERSION_MAJOR@.@VERSION_MINOR@.@VERSION_PATCH@
PROJECT_NUMBER = 1.0
``` 

# Code fragments are formated in a variable-width font

This problem can be fixed with small tweaks to the style sheet:

```diff
--- main.css	2021-02-03 23:20:40.000000000 +0100
+++ main.css	2021-02-04 23:54:25.000000000 +0100
@@ -7,7 +7,7 @@
 }

 h1, h2, h3, h4, h5, h6, p, a, li, span, blockquote, input, textarea, select, label {
-    font-family: 'Roboto', sans-serif !important;
+    font-family: 'Roboto', sans-serif;
 }

 p {

@@ -98,4 +100,8 @@
 }

-
+div.line,
+div.line a,
+div.line span {
+	font-family: monospace;
+}
```

JamesH65, 2021-02-10:
Removing brief was, IIRC, intentional in order to stop lots of duplicate lines when the source did not have a brief section. I'll take another look.

JamesH65, 2021-02-10:
(Thanks for the other comments  - vey useful, will also take a look at them. Thanks!)

JamesH65, 2021-02-11:
I'm not seeing the problem with the Brief descriptions on my build, but will add the JAVAAUTODOC anyway. The fixed space font fix works well, so will PR that, and have removed the extraneous tag. The PRODUCT_NUMBER issue had already been fixed as part of another change.

JamesH65, 2021-02-11:
Fixed by #117

cesarvandevelde, 2021-02-12:
Cheers, glad to see my remarks proved helpful! As a side-note, the Pico SDK docs are now also available as a [docset for Dash](https://github.com/Kapeli/Dash-User-Contributions/pull/3156). This is why I was trying to build the docs in the first place... :)

aallan, 2021-02-12:
> As a side-note, the Pico SDK docs are now also available as a docset for Dash. This is why I was trying to build the docs in the first place... :)

As a Mac user, that sounds good. Hadn't come across [Dash](https://kapeli.com/dash) before, but it looks interesting. How are the docsets updated when we make a new SDK release, is that a manual job?

cesarvandevelde, 2021-02-12:
I've created [a script](https://github.com/cesarvandevelde/pico-sdk-docset) that pulls in the pico SDK, runs doxygen, and packages everything up into a docset. The process is still manual in the sense that someone needs to run the script and submit a pull request to Dash, but most of the work is automated.

The docset Doxygen configuration is based on the one from the SDK, but some changes were made to improve the appearance in Dash (e.g. getting rid of the menubar).

lurch, 2021-02-12:
I keep meaning to ask - do you have any idea @cesarvandevelde why your groups-list screenshot looks so different to the one we generate [here](https://raspberrypi.github.io/pico-sdk-doxygen/modules.html) ?

JamesH65, 2021-02-12:
I suspect, but don't know, that the defaults for JAVADOC_BRIEF might be different?

cesarvandevelde, 2021-02-12:
@lurch Yup, it's caused by the briefs thing. As-is, the doxygen configuration does not pick up briefs for all those groups. I took a closer look at the output XML, and the brief tags are empty there as well. Adding `JAVADOC_AUTOBRIEF = YES` fixed the issue, causing Doxygen to [use the first line as brief](https://www.doxygen.nl/manual/config.html#cfg_javadoc_autobrief). Maybe the default value was changed in a recent Doxygen release?

closed 2021-05-31 by kilograham