raspberrypi/pico-sdk #96: pio.h: sm_config_set_sideset(): bad assertion

soundpaint, 2021-02-07:
pio.h says:

static inline void sm_config_set_sideset(pio_sm_config *c, uint bit_count, bool optional, bool pindirs) {
    assert(bit_count <= 32);
    c->pinctrl = (c->pinctrl & ~PIO_SM0_PINCTRL_SIDESET_COUNT_BITS) |
                 (bit_count << PIO_SM0_PINCTRL_SIDESET_COUNT_LSB);

However, according to RP2040 datasheet, table 397, SIDESET_COUNT occupies bits 31:29, that is only 3 bits, not 5.  That is, the maximum valid value is 7, not 32.

lurch, 2021-02-08:
Yeah, `PIO_SM0_PINCTRL_SIDESET_COUNT_BITS` is defined in `hardware/regs/pio.h` as `0xe0000000` which is `11100000000000000000000000000000` in binary, i.e. just the top 3 bits.

Fancy submitting a PR against the `develop` branch?

Wren6991, 2021-02-12:
Fixed by #105

closed 2021-02-12 by Wren6991