raspberrypi/pico-sdk #102: Calling  set_sys_clock_khz(438000, false)  hangs the Pico with 1.3V

Hermann-SW, 2021-02-08:
For code that determines maximal working overclocking for a single Pico I have currently to start with highest value of 437999 because of this bug.
Just did a run with 1.2V and it kept going down from 437999 until it reached 404000 and hangs.

lurch, 2021-02-08:
I guess you ought to expect undefined behaviour (e.g. crashes or hangs) at such high overclocks?
Or have I misunderstood and you're actually reporting something else?

Hermann-SW, 2021-02-08:
I expected to see whether setting and reading gpio works, not crashes.
I identified the reason for the crashes, I forgot to call set_sys_clock_48mhz() before doing printf() to USB minicom. So the reason for the hang seems to be printing at very high clocks to USB subsystem. Going fown to lower clocks before printing avoids the crash.

Wren6991, 2021-02-09:
>  I have currently to start with highest value of 437999 because of this bug.

What is the bug? You have described the hardware not functioning correctly at 3x its rated speed, and yes, this is why the rated speed is 3x lower than this. Please explain the **software** problem you would like to see fixed.

aallan, 2021-02-09:
The rated maximum clock for RP2040 is 133MHz. If you're overclocking you're entirely on your own. Unexpected behaviour like crashes should be expected.

closed 2021-02-09 by aallan