raspberrypi/pico-sdk #103: Inconsistent Function Names

soundpaint, 2021-02-08:
Various function names seem (at least to me) to have inconsistent names.
Examples:

- pio_sm_set_enabled() vs. pio_set_sm_mask_enabled():
  For the latter, I would have expected "pio_sm_set_mask_enabled()".
- pio_sm_restart() vs. pio_restart_sm_mask():
  For the latter, I would have expected "pio_sm_restart_mask()".
- pio_sm_clkdiv_restart() vs. pio_clkdiv_restart_sm_mask():
  For the latter, I would have expected "pio_sm_clkdiv_restart_mask()".
- pio_sm_claim() vs. pio_claim_sm_mask():
  For the latter, I would have expected "pio_sm_claim_mask()".
- pio_sm_unclaim() vs. pio_claim_unused_sm():
  For the latter, I would have expected "pio_sm_claim_unused()".

But maybe it is already too late for such changes, as possibly thousands of people have already started to make use of the API and expect stability?

Wren6991, 2021-02-09:
The first three examples are for manipulating global control groups of the PIO, not per-state machine controls, which is why the prefix is `pio_` and not `pio_sm_`

The `claim` thing is common across all interfaces using the `hardware_claim` library

I think there are multiple orthogonal things to be consistent with at once here.

lurch, 2021-02-09:
Just to expand slightly on Luke's comment... if you stare at it long enough, it does kinda make sense. It just depends how you break up the function-name into multiple parts:
 * pio_sm_set_enabled() vs. pio_set_sm_mask_enabled():
   * [pio_sm_set_enabled](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pio.html#gaa47ec4090dcdc251e44bf4e4cb89a7bc) - this does `set_enabled` for an individual `pio_sm_` and takes both `pio` and `sm` parameters
   * [pio_set_sm_mask_enabled](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pio.html#gaf726f77ee21f5373cadf064a4e0b4935) - this does `set_sm_mask_enabled` for a `pio_` and has a `pio` parameter but not a `sm` parameter
 * pio_sm_restart() vs. pio_restart_sm_mask():
   * [pio_sm_restart](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pio.html#ga7b89b93dd0553c77538c70baf6a7e4f0) - this does `restart` for an individual `pio_sm_` and takes both `pio` and `sm` parameters, i.e. it operates at the level of a state-machine
   * [pio_restart_sm_mask](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pio.html#gae3e115fd41d9ac29c7a13b0c213bc2cd) - this does `restart_sm_mask` for a `pio_` and has a `pio` parameter but not a `sm` parameter, i.e. it operates at the level of a pio. If it was instead named `pio_sm_restart_mask` as you suggest, then using these same naming conventions would imply that it was trying to `restart_mask` at the state-machine level, which doesn't make sense?
   * etc.
   * pio_sm_unclaim() vs. pio_claim_unused_sm():
     * [pio_sm_unclaim](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pio.html#ga62328112de918932e4ed0955ce0054ab) - has `pio` and `sm` parameters
     * [pio_claim_unused_sm](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pio.html#ga604c071438065c46c9f4f21bdb61befd) - only has a `pio` parameter

Just to play devil's advocate, I guess the functions _could_ have instead been named:
 * pio_set_sm_enabled() and pio_set_sm_mask_enabled()
 * pio_restart_sm() and  pio_restart_sm_mask()
 * etc. etc.

Given the inherent complexity here (and my lack of familiarity with the PIO API), I'm not going to try and argue for one naming-scheme over another! But (when viewed from the right angle) things aren't as inconsistent as you claim they are :slightly_smiling_face:

soundpaint, 2021-02-10:
Sorry for my sporadically high response latency!
Ok, I think now I understand the reasoning behind your naming scheme; still, it _feels_ somehow strange.  But I would like to withdraw my naming suggestion, since I agree that mine is not at all better, when considering the whole picture. :-)

lurch, 2021-02-10:
>  I would like to withdraw my naming suggestion

So the issue can be closed now? :slightly_smiling_face:

soundpaint, 2021-02-11:
> > I would like to withdraw my naming suggestion
> 
> So the issue can be closed now?

I guess so.  While I am still not really happy, I do not see how to make it better...

closed 2021-02-11 by soundpaint

lurch, 2021-02-11:
If you're that unhappy you could always make you own `soundpaint_pio_functions.h` as a wrapper on top of `hardware/pio.h` ? :wink: