raspberrypi/pico-sdk #99: No CMAKE_CXX_COMPILER could be found

littleningmeng, 2021-02-08:
ELF2UF2 will need to be built
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/apache/github/pico-sdk
apache@ubuntu20:~/github/pico-sdk$ make
[  0%] Built target bs2_default
[  0%] Performing configure step for 'ELF2UF2Build'
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/apache/github/pico-sdk/elf2uf2/CMakeFiles/CMakeOutput.log".
See also "/home/apache/github/pico-sdk/elf2uf2/CMakeFiles/CMakeError.log".
make[2]: *** [test/pico_stdlib_test/CMakeFiles/ELF2UF2Build.dir/build.make:107：test/pico_stdlib_test/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure]

lurch, 2021-02-08:
Could you please explain which set of instructions you've been following, i.e. which steps you ran before this one, and on which platform / OS you're running?
I see you have `Build files have been written to: /home/apache/github/pico-sdk` in your output, whereas all our instructions would result in your Build files being in a `pico-sdk/build` subdirectory...

littleningmeng, 2021-02-09:
```bash
pache@apache-vm:~/github$ cd pico-sdk/
apache@apache-vm:~/github/pico-sdk$ ls
cmake           docs      lib          pico_sdk_init.cmake     README.md  test
CMakeLists.txt  external  LICENSE.TXT  pico_sdk_version.cmake  src        tools
apache@apache-vm:~/github/pico-sdk$ mkdir build
apache@apache-vm:~/github/pico-sdk$ cd build/
apache@apache-vm:~/github/pico-sdk/build$ cmake ..
PICO_SDK_PATH is /home/apache/github/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/apache/github/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.8 (found version "3.8.2") found components: Interpreter 
CMake Warning at src/rp2_common/tinyusb/CMakeLists.txt:10 (message):
  TinyUSB submodule has not been initialized; USB support will be unavailable

           hint: try 'git submodule update --init'.

ELF2UF2 will need to be built
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/apache/github/pico-sdk/build
apache@apache-vm:~/github/pico-sdk/build$ cd ..
apache@apache-vm:~/github/pico-sdk$ git submodule update --init
子模组 'tinyusb'（https://github.com/raspberrypi/tinyusb.git）已对路径 'lib/tinyusb' 注册
正克隆到 '/home/apache/github/pico-sdk/lib/tinyusb'...
子模组路径 'lib/tinyusb'：检出 'e0aa405d19e35dbf58cf502b8106455c1a3c2a5c'
apache@apache-vm:~/github/pico-sdk$ cd build/
apache@apache-vm:~/github/pico-sdk/build$ cmake ..
PICO_SDK_PATH is /home/apache/github/pico-sdk
PICO platform is rp2040.
PICO compiler is 
PICO_GCC_TRIPLE defaulted to arm-none-eabi
PICO target board is pico.
Using board configuration from /home/apache/github/pico-sdk/src/boards/include/boards/pico.h
TinyUSB available at /home/apache/github/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
ELF2UF2 will need to be built
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/apache/github/pico-sdk/build
apache@apache-vm:~/github/pico-sdk/build$ make
Scanning dependencies of target bs2_default
[  0%] Building ASM object src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/boot2_w25q080.S.obj
[  0%] Linking ASM executable bs2_default.elf
[  0%] Built target bs2_default
Scanning dependencies of target ELF2UF2Build
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] No download step for 'ELF2UF2Build'
[  1%] No patch step for 'ELF2UF2Build'
[  1%] No update step for 'ELF2UF2Build'
[  1%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is unknown
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/apache/github/pico-sdk/build/elf2uf2/CMakeFiles/CMakeOutput.log".
See also "/home/apache/github/pico-sdk/build/elf2uf2/CMakeFiles/CMakeError.log".
make[2]: *** [test/pico_stdlib_test/CMakeFiles/ELF2UF2Build.dir/build.make:107：test/pico_stdlib_test/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-configure] 错误 1
make[1]: *** [CMakeFiles/Makefile2:1681：test/pico_stdlib_test/CMakeFiles/ELF2UF2Build.dir/all] 错误 2
make: *** [Makefile:84：all] 错误 2
```

littleningmeng, 2021-02-09:
Hi @lurch 
My system is Ubuntu20.04LTS
cmake version is 3.16.3

kilograham, 2021-02-09:
looks like you are mising a native C++ compiler;

perhaps `sudo apt install g++` or `sudo apt install build-essentials`

littleningmeng, 2021-02-09:
I have already installed g++ and build-essential
Maybe not this reason

lurch, 2021-02-09:
Thanks for the detailed logs :+1: 
I see your hostname is `apache-vm` - are you using Docker or something similar? Sometimes these run stripped-down versions of the OS which behave differently to the vanilla OS versions. (different packages installed, different PATH settings, etc.)

What do each of these commands report:
```
gcc --version
g++ --version
which gcc
which g++
echo $PATH
```

littleningmeng, 2021-02-09:
I'am using Ubuntu20.04LTS on VMware Workstation Pro
```bash
apache@apache-vm:~/tmp/helloworld/dist$ which gcc
/usr/bin/gcc
apache@apache-vm:~/tmp/helloworld/dist$ which g++
/usr/bin/g++
apache@apache-vm:~/tmp/helloworld/dist$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

apache@apache-vm:~/tmp/helloworld/dist$ g++ --version
g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

apache@apache-vm:~/tmp/helloworld/dist$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
apache@apache-vm:~/tmp/helloworld/dist$ uname -a
Linux apache-vm 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
apache@apache-vm:~/tmp/helloworld/dist$
```

lurch, 2021-02-09:
Hmmm, strange that `cmake` can't find `g++` then, given that it _is_ in the PATH.
Have you tried doing `CXX=g++ cmake ..` (from within the `build/`directory) as the cmake error suggests?

lurch, 2021-02-09:
@kilograham Would it make sense (or be possible?) to do this check when the user first runs `cmake`, so they see the error before they later run `make` ?

littleningmeng, 2021-02-09:
> Hmmm, strange that `cmake` can't find `g++` then, given that it _is_ in the PATH.
> Have you tried doing `CXX=g++ cmake ..` (from within the `build/`directory) as the cmake error suggests?

This works!
Thanks a lot ;) 
`cd build && CXX=g++ cmake .. && make`
``bash
[100%] Linking CXX executable hardware_pwm_test.elf
[100%] Built target hardware_pwm_test
```

closed 2021-02-09 by littleningmeng

pflannery, 2021-06-24:
@lurch is there a docker image with the dev tools already setup?

lurch, 2021-06-24:
@pflannery I dunno. It's not something we've done ourselves; but I dunno if a community-member might have done that, as it's not something that I've looked for.
If  you're having problems with the instructions in https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf please open a new issue in https://github.com/raspberrypi/pico-feedback