raspberrypi/pico-sdk #104: Multiple missing 'extern "C"'

geurtv, 2021-02-09:
The following header files are all missing __cplusplus ifdef'ed 'extern "C"'

src/rp2_common/hardware_claim/include/hardware/claim.h
src/rp2_common/hardware_divider/include/hardware/divider.h
src/rp2_common/hardware_flash/include/hardware/flash.h
src/rp2_common/hardware_rtc/include/hardware/rtc.h
src/rp2_common/hardware_watchdog/include/hardware/watchdog.h
src/rp2_common/pico_bootrom/include/pico/bootrom.h
src/rp2_common/pico_stdio_semihosting/include/pico/stdio_semihosting.h
src/rp2_common/pico_stdio_uart/include/pico/stdio_uart.h
src/rp2_common/pico_stdio_usb/include/pico/stdio_usb.h

Also the following header-only files ... might want to add it for consistency:

src/rp2_common/hardware_base/include/hardware/address_mapped.h
src/rp2_common/hardware_resets/include/hardware/resets.h

Wren6991, 2021-02-09:
Thank you for giving the exhaustive list! Made things quicker.

Wren6991, 2021-02-11:
Fixed by #106

commit b8d23411 (2021-02-09): Add missing extern "C" to header files (closes #104)
commit b34995a5 (2021-02-11): Add missing extern "C" to header files (closes #104)
closed 2021-02-11 by Wren6991