raspberrypi/pico-sdk #108: calling `sleep_ms` inside an interrupt handler causes lockup

Lard4, 2021-02-10:
Summary: when you call `sleep_ms` inside of an interrupt handler, some sort of infinite loop occurs inside of the hardware timer. Specifically, picoprobe/gdb is pointing to the code hanging somewhere in this function: https://github.com/raspberrypi/pico-sdk/blob/26653ea81e340cacee55025d110c3e014a252a87/src/rp2_common/hardware_timer/include/hardware/timer.h#L109

To reproduce: build/flash https://github.com/Lard4/pico-irq-test then when GP22 goes high (gets the interrupt), it turns off the LED then hangs.

lurch, 2021-02-10:
Looking at https://github.com/raspberrypi/pico-sdk/blob/master/src/common/pico_time/time.c#L332 it seems `sleep_ms` calls `sleep_us`, which in turn calls `sleep_until` which relies on an alarm-interrupt; so perhaps the problem is interrupts being fired from inside interrupts? :shrug: 
If you _must_ use `sleep_ms` inside an interrupt handler (which seems like an odd thing to do, because you typically want your interrupts to exit as quickly as possible?), it looks like you can set `PICO_TIME_DEFAULT_ALARM_POOL_DISABLED` to 1, and it'll then do the sleep using busy-waits instead of alarm IRQs?

Ahhh, and I've just noticed that https://raspberrypi.github.io/pico-sdk-doxygen/group__sleep.html says "These functions should not be called from an IRQ handler." :slightly_smiling_face:

Wren6991, 2021-02-10:
Yes, `sleep_` functions are interrupt-driven (to save power) and you are using them inside an interrupt. Sleeping in an IRQ is generally unsupported on *any* platform, and the function description tells you not to do this. Having long delays in an IRQ is also pretty unfriendly to the code being interrupted.

If you do want to insert a long delay in an IRQ, the function to use is `busy_wait_us_32()` which you will see is also listed as a "see also" under the sleep functions.

[image]

[image]

kilograham, 2021-02-12:
I think nonetheless we should downgrade a sleep in an IRQ handler to a busy_wait_us automatically (by default)

swetland, 2021-02-14:
I'd recommend leaving sleep disallowed from IRQ context (as is the case in just about every OS out there), and perhaps instead add an assert(irqs-are-enabled) in sleep in debug builds.  The chances of waiting any amount of time in irq context ever being the right thing to do are near zero.

kilograham, 2021-02-14:
yeah perhaps i was feeling overly generous

kilograham, 2021-02-27:
merged into develop - sleep in an IRQ handler remains disallowed

closed 2021-02-27 by kilograham