raspberrypi/pico-sdk #113: watchdog_caused_reboot() returning true when going from BOOTSEL -> normal boot

maxnet, 2021-02-10:
Currently watchdog_caused_reboot() also returns true if you just flashed an image using the BOOTSEL mode method, and it restarts to the program afterwards.

Technically, this may be correct, as you probably use the watchdog to reboot after bootsel did its thing.
However the question most people calling such function actually want the answer of is "did watchdog cause reboot while running my main program ?" 

Guess I can include "hardware/structs/watchdog.h" and set some magic value to scratch register myself to indicate my main program started. And test for that in addition to watchdog_caused_reboot(), but may be better to have a standard method for such thing.

Wren6991, 2021-02-12:
> Technically, this may be correct, as you probably use the watchdog to reboot after bootsel did its thing.

Yes that's right.. In 2.8.2.1 you can see the bootrom feature that is used to vector through the watchdog like this:

[image]

You can see where the bootrom checks for a vector here: https://github.com/raspberrypi/pico-bootrom/blob/00a4a19114195e20fb817bdfbca1165e157eef37/bootrom/bootrom_rt0.S#L266

So a test like `SCRATCH5 + SCRATCH7 + 0xb007c0deu == 0u` will let you detect this type of watchdog event in your startup code, as long as you then clear those scratch registers afterward so that you can disambiguate from your other watchdog events.

maxnet, 2021-02-12:
>So a test like (SCRATCH5 ^ SCRATCH7) + 0xb007c0deu == 0u will let you detect this type 
>of watchdog event in your startup code

Yes, just zeroing scratch5 and testing for zero seems to be good enough to detect the case of the watchdog being triggered in my main code.

```
    if (watchdog_caused_reboot() && watchdog_hw->scratch[5] == 0)
    {
        // handle hang in main code on previous boot
    }
    watchdog_hw->scratch[5] = 0;    
    watchdog_enable(1000, 1);
```

Still wonder if this doesn't belong in the sdk by default.
Rather than me having to access low level scratch register, and hoping they will not change in a future chip release.

As in always zero scratch5 in watchdog_enable() and test for it being zero in watchdog_caused_reboot() (or add a watchdog_caused_reboot_in_main_code() )

Wren6991, 2021-02-12:
> Still wonder if this doesn't belong in the sdk by default.
Sounds like a good idea to me

> As in always zero scratch5 in watchdog_enable() and test for it being zero in watchdog_caused_reboot() (or add a watchdog_caused_reboot_in_main_code() )

Probably the latter, to avoid stomping on other uses of those scratch registers

closed 2021-10-20 by kilograham