raspberrypi/pico-sdk #114: Documentation: pio_sm_clkdiv_restart(): Misleading Description

soundpaint, 2021-02-11:
The documentation says on pio_sm_clkdiv_restart():

"Restart a state machine's clock divider (resetting the fractional count)"

This sounds, like _only_ the fractional part would be reset, if the phrase in parentheses is taken as an explanation for the phrase before.
In fact, the integer count value is reset _and_ the fractional accumulator (this is not a count!) is cleared, according to RP2040 datasheet, Sect. 3.7, Table 379.

Same for pio_clkdiv_restart_sm_mask().

soundpaint, 2021-02-11:
Thank you, sounds great!  Just a single nitpick:

> "This function clears the divider's integer and fractional phase accumulators"

I would have expected "accumulator" (singular form), since the integer part is (in contrast to the fractional part) used as a counter, at least as far as I understand.  (The integer part is increased by one whenever the accumulator overflows, but otherwise solely acts as a decreasing counter.  In contrast, upon each clock cycle, the fractional divider part is added to the accumulator (thus effectively accumulating the summed up fractional time) and wraps around upon overflow). :-)

Wren6991, 2021-02-11:
Whether a downcounter can be considered an accumulator where only -1 is added, in the digital design sense, is a philosophical question that is beyond me -- what I am certain of is that it accumulates phase :)

Thanks for raising this by the way, there were some other things in there that needed clearing up.

commit 0fbaaabe (2021-02-11): pio.h doxygen fixes and clarifications (closes #114)
closed 2021-02-11 by Wren6991