raspberrypi/pico-sdk #120: Raspberry Pi Pico not showing up as COM/ttyACM0 device

Noxdor, 2021-02-11:
I just started working with the new Pico and run into some problems while trying to communicate with it over Serial UART.

The Pico doesn't show up neither as a com port on Windows nor as a ttyACM0 device in my ubuntu subsystem for Windows.
I tried it with two cables and two different Picos.
Then I checked if the cables were fine by plugging in an ESP32 instead. It was immediately recognized as a COM Port on Windows. I couldn't try it on the subsystem, since it stil doesn't show up as a ttyACM0 device. The article
https://www.scivision.dev/usb-tty-windows-subsystem-for-linux/
might suggest though, that these kind of devices are renamed in the subsystem.

Anyways, the Pico clearly doesn't show up as a COM Port! That makes it impossible to communicate via Serial. Anyone having an explanation or a fix?

Wren6991, 2021-02-11:
> I just started working with the new Pico and run into some problems while trying to communicate with it over Serial UART

Serial UART is a pair of GPIOs for RX and TX (GPIO0 and GPIO1 by default). Do you mean UART, or do you mean USB CDC?

You need to enable the USB CDC in your code (it's not enabled by default as you may want to use USB for other things). The `hello_usb` example has it enabled. If you program that example, do you now see Pico appearing as a USB serial device?

Noxdor, 2021-02-12:
Great explanation, you were absolutely right, problem fixed ;.)

closed 2021-02-12 by Noxdor

panchalnikunj, 2021-02-13:
https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ

Once visit this channel for Raspberry Pi Pico Projects