raspberrypi/pico-sdk #124: Locally compiled hello_usb does not work.

BenoitVII, 2021-02-12:
Hello, I tried to compile locally the pico-examples project following the C++ guide on Windows.
I managed to compile it without any error and I have a hello_usd.uf2 file generated inside my /build folder.

However when I upload it on the pico, **no serial port appears**. 

It works fine with the downloaded hello_usb.uf2 file from the quick start guide on the Raspberry website.

**There is no error showing up during the compilation,** and I noticed that the size of the 2 binaries are different : **25ko** for my compiled version and **45ko** for the downloaded version.

Any idea?

**Compilation machine** : Windows 10, intel processor.

**Python 3.8 instead of 3.7** but I think this has no impact.

**SDK Window 10 v.10.0.17763.0**

aallan, 2021-02-12:
Did you do

```
$ git clone -b master https://github.com/raspberrypi/pico-sdk.git 
$ cd pico-sdk
$ git submodule update --init
$ cd ..
$ git clone -b master https://github.com/raspberrypi/pico-examples.git
```
do an `submodule update --init` inside the `pico-sdk` directory? If not then the `tinyusb` module won't be pulled into the build and you won't have any USB support, so no USB serial port.

BenoitVII, 2021-02-12:
You are right, @aallan , thank you.
I was actually lazy to do a git clone so I downloaded the source directly in a zip file and skipped this step.

Maybe it would be nice to have an error instead of a warning for this feature since it's something that a lot of people want?

kilograham, 2026-11-11:
i think the warning is fine

closed 2026-11-11 by kilograham