raspberrypi/pico-sdk #127: Mechanism for setting exception handlers

davidgiven, 2021-02-13:
I could be wrong, but there doesn't appear to be any way of setting exception handlers from code other than to create specially-named functions such as `isr_svcall`, which the documentation suggestsnot to do.

The implementation numbers interrupts/exceptions from 0 for IRQ0. As these appear immediately after the vector table, it ought to be possible to use negative numbers to refer to the exceptions --- these are actually documented in the ARM documentation at https://developer.arm.com/documentation/dui0497/a/the-cortex-m0-processor/exception-model/exception-types, so `SVCall` is interrupt -5. However, calling `irq_set_exclusive_handler()` doesn't work because you can only use it on vectors which are already pointing at `__unhandled_user_irq`.

Given how clean the rest of the SDK's design is, it'd be nice to have a blessed-by-the-API way of doing this.

kilograham, 2021-02-14:
that is really good point! you can certainly use isr_svcall as a workaround for now, but we should add an API

kilograham, 2021-04-01:
note also that the vector table is installed in RAM by default, also, so you could write to this... 

Also note, that we are working on FreeRTOS "port"

kilograham, 2021-05-12:
merged into develop

PR #380 (2021-05-05): Add hardware_exception for setting exception handlers at runtime
fixes #127
closed 2021-05-12 by kilograham