raspberrypi/pico-sdk #128: Wstrict-prototypes

Wiz-IO, 2021-02-14:
Hi,

please correct **void** prototypes in header files...

as example:
src\common\pico_stdlib\include/pico/stdlib.h:74:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]

kilograham, 2021-02-14:
already done in develop

closed 2021-02-14 by kilograham

lurch, 2021-02-14:
(fixed by #95 )

Wiz-IO, 2021-03-08:
[hardware/sync.h](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_sync/include/hardware/sync.h#L208):208:54: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

lurch, 2021-03-08:
@Wiz-IO It's probably best to report problems with `-Wsign-compare` in a new issue, rather than adding them to a **closed** issue about `-Wstrict-prototypes` :wink: