raspberrypi/pico-sdk #131: `hardware_alarm` functions on wrong page

davidgiven, 2021-02-15:
Something isn't right with the documentation for the `hardware_alarm` functions. The `timer.h` page (https://raspberrypi.github.io/pico-sdk-doxygen/rp2__common_2hardware__timer_2include_2hardware_2timer_8h.html#a62098633d173c24c56c359df2e886ed2) documents _only_ `hardware_alarm_cancel()` and `hardware_alarm_set_target()`, but none of the others. Meanwhile the hardware_timer page (https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__timer.html#gaa4c3c72008c100ca25f94b806f13f112) documents all the functions _except_ `hardware_alarm_cancel()` and `hardware_alarm_set_target()`.

I think that these two functions are incorrectly tagged somehow so that they appears in `timer.h`'s documentation but not in `hardare_timer'.

kilograham, 2021-02-17:
will move them to the same group as the others

kilograham, 2021-02-18:
merged to master

commit ffc7f19f (2021-02-18): move two hardware_timer methods to the right doxygen group (fix #131)
closed 2021-02-18 by kilograham