raspberrypi/pico-sdk #133: Problems with cmake compilation (pico/stdlib.h: No such file or directory)

miki164, 2021-02-15:
Hi
During going through READMEs "Quick-start your own project". I encountered the error 
```
pico/stdlib.h: No such file or directory
```
after running 
```
make hello_world
```
I solved the problem running the cmake not in build folder but in main folder of project(leaving all generated files there).
Is there a better solution to that problem?
Cmake version 3.16.3
OS Linux Ubuntu 20.10

ndinsmore, 2021-02-16:
@miki164 I have run ingot a similar problem on OSX and for me the solution is to delete and recreate the "build" directory

miki164, 2021-02-16:
I've tried that and doesn't work for me.

lurch, 2021-02-17:
Ahh... I think I see the problem. A couple of the steps were in the wrong order, which might lead to you creating files in the wrong directory.

Does #142 fix things @miki164 ?

miki164, 2021-02-17:
Yes, it works. Thank you for help

closed 2021-02-17 by miki164

antalpeter, 2023-08-23:
Is there any simple way to translate:

#include <gpio.h>
#define PICO_DEFAULT_LED_PIN 25

int main() {
#ifndef PICO_DEFAULT_LED_PIN
#warning blink example requires a board with a regular LED
#else
    const unsigned int LED_PIN = PICO_DEFAULT_LED_PIN;
    gpio_init(LED_PIN);
    gpio_set_dir(LED_PIN, 1);
    while (1) {
        gpio_put(LED_PIN, 1);
        sleep_ms(250);
        gpio_put(LED_PIN, 0);
        sleep_ms(250);
    }
#endif

lurch, 2023-08-23:
> Is there any simple way to translate:

What do you mean by "translate"? :confused: 

This issue-tracker is for reporting bugs; if you just have questions then https://forums.raspberrypi.com/viewforum.php?f=143 is a more suitable place for that.

PR #142 (2021-02-18): README.md improvements
Fix the order of the steps (fixes #133) and make it clearer that step 2 is choose-just-one-option
commit 9fbfa1fc (2021-02-18): REAMDE.md improvements

Fix the order of the steps (fixes #133) and make it clearer that step 2 is choose-just-one-option