raspberrypi/pico-sdk #134: Build Compile Warning - PICO_FLASH_SIZE_BYTES redefined when setting this value in a board definition file.

kirk-sfe, 2021-02-15:
Our rp2040 boards have 16MB on-board Flash, and we are defining this about in the board definition include file via the PICO_FLASH_SIZE_BYTES macro. 
i.e:
// board has 16M onboard flash
#define PICO_FLASH_SIZE_BYTES (16 * 1024 * 1024)

But when building a project, this board definition conflicts the value being set in:
../src/rp2040/hardware/regs/platform_defs.h

The pre-processor spits out a warning: "note: this is the location of the previous definition"

Which has the value set to:
#ifndef PICO_FLASH_SIZE_BYTES
#define PICO_FLASH_SIZE_BYTES (2 * 1024 * 1024)
#endif

This seems like a bug, OR we're using this #define incorrectly in our board definition. 

Note - not a critical issue - everything still works, just an annoying message during project build.

kilograham, 2021-02-15:
yes that is a bug; it should not be set before the user/board configuration includes!

lurch, 2021-03-03:
Fixed in #159 , so this issue can be closed.

closed 2021-03-05 by kilograham