raspberrypi/pico-sdk #141: Multiple rising or falling edges detected in a row with gpio_set_irq_enabled_with_callback

BenoitVII, 2021-02-17:
Hello guys,

I'm currently trying to use external pin interrupts with the pico. I try to detect both rising and falling edges, and my issue is that the result is not consistent. My goal is to read a rotary encoder input so I attached GPIO10 to pinA of my encoder.

I expect to have a perfect "rising / falling / rising / falling / ... " pattern, as the pin will be alternatively pulled either high or low while I turn the encoder (it should be the case with all kind of inputs, as the pin will always go from **high** to _low_ and from _low_ to **high**, not from **high** to **high** or from _low_ to _low_, which doesn't make any sense).

However, i'm not getting that, and **multiple rising or falling edges are detected in a row** as you can see with the putty output:
[image]
(You can spot sometimes I have 2 or 3 "0"s or "1"s corresponding to falling or rising edges).

Bellow is the code producing this. I retrieved the gpio_event_string function from the pico-example repository, and adapted the output to make it clearer.

```c++
#include <stdio.h>
#include "pico/stdlib.h"

static char event_str[128];

void gpio_event_string(char *buf, uint32_t events);

//callback for rising/falling edge.
void gpio_callback(uint gpio, uint32_t events)
{
    //get a string of what happened during this callback
    gpio_event_string(event_str, events);
    printf("%s", event_str);
}

int main()
{
    //init serial for debug
    stdio_init_all();

    // use pull-up
    gpio_init(10);
    gpio_pull_up(10);
    // register callback on both rising and falling edges.
    gpio_set_irq_enabled_with_callback(10, GPIO_IRQ_EDGE_RISE | GPIO_IRQ_EDGE_FALL, true, &gpio_callback);
    
    while(1){
        //do nothing.
    }
}

static const char *gpio_irq_str[] = {
    "LEVEL_LOW",  // 0x1
    "LEVEL_HIGH", // 0x2
    "0",  // 0x4 edge fall
    "1"   // 0x8 edge rise
};

//retrieved from pico-example
void gpio_event_string(char *buf, uint32_t events)
{
    for (uint i = 0; i < 4; i++)
    {
        uint mask = (1 << i);
        if (events & mask)
        {
            // Copy this event string into the user string
            const char *event_str = gpio_irq_str[i];
            while (*event_str != '\0')
            {
                *buf++ = *event_str++;
            }
            events &= ~mask;
        }
    }
    *buf++ = '\0';
}
```

T=hank you!

lurch, 2021-02-17:
Possibly related to debouncing? See https://github.com/raspberrypi/pico-examples/pull/45

lurch, 2021-02-17:
Also, `printf` can use quite a lot of CPU cycles so it's possible that you "missed" a rising or falling edge whilst the CPU was busy doing printf inside your interrupt-handler.

BenoitVII, 2021-02-17:
Thank you lurch. Indeed maybe I can try to reduce the duration of my interrupt function for my testings.

By the way, I didn't find any documentation regarding the way interrupt data is send back in the `gpio_irq_callback_t ` method.
It's an uint8_t with groups of 4 bits representing the type of interruption (rising, falling, ...) but I don't understand how that can happen simultaneously on one pin (sometimes I get 0011 meaning that the interrupt got both rising AND falling). Any insight on that?

lurch, 2021-02-17:
There's a bit of documentation at https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__gpio.html#ga594e390dc5e7407a35bcce6cae73ae2c which says "Currently the GPIO parameter is ignored, and this callback will be called for any enabled GPIO IRQ on any pin." so maybe you had a rising edge on one pin at the same time as falling on another pin? :shrug: 

ping @kilograham The doxygen documentation seems to be missing a description (or even a declaration) of the `gpio_irq_callback_t` type?

BenoitVII, 2021-02-17:
I check the `gpio` parameter in my `gpio_callback`, only the pin 10 is triggering interrupts.
I tried to implement the debouncing as in the PR you shared, and still get quite weird results (each line corresponds to an interrup). This is while trying to ground and release manually the gpio.
```

10 -> EDGE_FALLEDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_FALLEDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_FALLEDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_FALLEDGE_RISE
10 -> EDGE_FALLEDGE_RISE
10 -> EDGE_FALLEDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_FALLEDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_FALLEDGE_RISE
```

Interestingly enough, I tried another thing. In the `main` function i create a square output with a period of 10ms, and the results become perfect

```

10 -> EDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_RISE
10 -> EDGE_FALL
10 -> EDGE_RISE
10 -> EDGE_FALL
```

lurch, 2021-02-18:
Yes, whenever you do things "manually" (i.e. switches, buttons, rotary encoders, jumper wires, etc.) you'll get contact bounce. Which you can either fix "in software" by adding delays etc. or you can fix "in hardware" by adding a capacitor.

BenoitVII, 2021-02-18:
I agree but even with debouncing the results I get make no sense to me. I really don’t understand why sometimes we get both edge directions in one irq even when I debounce.. Any documentation on this would be really appreciated

lurch, 2021-02-21:
> The doxygen documentation seems to be missing a description (or even a declaration) of the gpio_irq_callback_t type?

I just created #175 to cover this.

lurch, 2021-03-02:
I've just been looking into this in a bit more detail, and the RP2040 datasheet says:
"The level interrupts are not latched. This means that if the pin is a logical 1 and the level high interrupt is active, it will
become inactive as soon as the pin changes to a logical 0. The edge interrupts are stored in the INTR register and can be
cleared by writing to the INTR register."

The `INTR` register shows the "raw" interrupts seen by the pin, and the edge interrupts _are_ latched, so even if using software debouncing you'd "see" both a rising and falling edge reported to your callback if both those transitions (i.e. a blip) happened between each time your callback got called. Off the top of my head (bear in mind I'm more of a software-guy than a hardware-guy), I guess you could either add sufficient hardware debouncing so that rapid edge-blips don't get "seen" by the pin, or you could instead monitor the previous-state and current-state of the LEVEL_LOW and LEVEL_HIGH signals, and use those to work out whether you had a rising or falling transition?

BenoitVII, 2021-03-02:
Hello, I don't really understand what latching means in this context (English is not my motherthong, which probably don't help either). Your ideas are both good, and actually I even decided to go without interrupts and just poll the state of the pins in my main loop.

The fact is that it's the first time I see this behaviour on a microprocessor. I'm not very experienced but I worked quite a lot with arduinos and ESP32, and each time I used pin interrupts, it registered a sequence of falling/rising/falling, without any hardware debouncing. Note that the arduino clock is a lot slower than RP2040's, so I think this is not a problem of bouncing beeing so fast that it's not registered by the microprocessor. 

This is why I think that either it is something that I don't understand about RP2040 interrupts, or there really is a bug in sdk.

I will try this week to use interrupts on the micropython sdk to see how it is implemented there.

lurch, 2021-03-02:
"latching" means that it remembers "what it saw" until you explicitly clear it.

liamfraser, 2021-03-02:
@BenoitVII how are you debouncing? Unless you have some external hardware to do the debouncing then the EDGE_HIGH and EDGE_LOW registers are basically connected straight to the pin. So if the pin is floating at any point then it makes sense that you could see both EDGE_HIGH and EDGE_LOW events. Even if the pin has a pull up or pull down that will take some time to take effect.

BenoitVII, 2021-03-02:
@lurch Ok thanks for the explanation. 
@liamfraser I'm not debouncing with anything. I activated the internal pullup for the pin so it's never  really floating.

liamfraser, 2021-06-01:
Closing this as it appears to be debounce related.

closed 2021-06-01 by liamfraser