raspberrypi/pico-sdk #162: absolut_time_t prints 0

akhilbhas, 2021-02-19:
HI I have the following code snippet
`
absolute_time_t temp_time;
absolute_time_t prev_time
void gpio_callback(uint gpio, uint32_t events) {
        temp_time = get_absolute_time();
        printf("GPIO %d tmp:%d \n",gpio,temp_time);
}

int main()
{
   stdio_init_all();
    gpio_pull_up(2);
    gpio_set_irq_enabled_with_callback(2, GPIO_IRQ_EDGE_FALL, true, &gpio_callback);
    sleep_ms(1000);
    prev_time = get_absolute_time();
    while(true){
        sleep_ms(100);
        prev_time = get_absolute_time();
        printf("prev:%d \n",prev_time);
    }
`

for the print in the while loop, I am always getting 0. But in the call back I am getting value. Not able to figure out the issue
Please someone help

lurch, 2021-02-19:
I think [absolute_time_t](https://raspberrypi.github.io/pico-sdk-doxygen/structabsolute__time__t.html) might be an "opaque type"? You might get more reliable results if you use [to_ms_since_boot()](https://raspberrypi.github.io/pico-sdk-doxygen/group__timestamp.html#gab12467c48bde27171b552ac4dc8c7d59) to convert it to a `uint32_t`?

akhilbhas, 2021-02-19:
@lurch thanks for the information, I have never come across opaque type. Let me research a little bit and implement.

closed 2021-02-19 by akhilbhas