raspberrypi/pico-sdk #163: Missing GPIO out read function in SDK

Volksolive, 2021-02-19:
I haven't found in the SDK datasheet and hardware/gpio a function to read the GPIO output register.
Also in the hardware struct sio.h, the out register is mentioned as write only (io_wo_32 gpio_out;) but I think should be read write.

In the optic, the function gpio_get_pad() give the INFROMPAD info but there is no function to get the OUTTOPAD.

A side note on the RP2040 datasheet, the pads register are quite far down in the list of registers section, maybe it would be nice to add a reference to it (in the signets, or at the beginning of the section 2.19.6) so that it would be easier to access it instead of scrolling tens of pages.

lurch, 2021-02-19:
> I haven't found in the SDK datasheet and hardware/gpio a function to read the GPIO output register.

Note that https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/hardware_regs/include/hardware/regs/sio.h#L49 says "Reading back gives the last value written, NOT the input value from the pins." What's your use-case?

> Also in the hardware struct sio.h, the out register is mentioned as write only (io_wo_32 gpio_out;) but I think should be read write.

Yeah, https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/hardware_regs/include/hardware/regs/sio.h#L64 says `RW` so _perhaps_ https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2040/hardware_structs/include/hardware/structs/sio.h#L20 should be `io_rw_32` ? That's a question for @kilograham / @Wren6991 
(seems to be quite a few registers which are `RW` in hardware/regs/sio.h but `io_wo_32` in hardware/structs/sio.h)

> In the optic, the function gpio_get_pad() ...

@kilograham `gpio_get_pad` is declared at https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_gpio/gpio.c#L18 but doesn't have an entry (and therefore no doxygen) in https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_gpio/include/hardware/gpio.h - is that intentional?

> give the INFROMPAD info but there is no function to get the OUTTOPAD.

Again, an example use-case might be helpful?

Volksolive, 2021-02-19:
The use case is to be able to read what was written in the out register.
in that case you don't need to have a variable that is a copy of the out register and it might be easier when working with 2 cores to know what the other core have set the output to so that you don't have share a variable between the 2 cores and I believe the read to the register is atomic.

for the OUTTOPAD, reading the pad directly can remove the need to apply some complement logic to the out value if the OUTOVER is on. but also not having to know which peripheral manage this pin, just getting its out state.
I mean it is the same way as you would read input pad instead of the SIO (gpio) input register.

On another note, IMO it is quite disturbing that the GPIO Out, In, etc, registers are listed in the SIO section, I understand why but my first reflexes would be to check the GPIO section to learn how to read or write to a GPIO. maybe if the reference to the SIO section would be more highlighted in the GPIO section (only one reference to it in the overview section) maybe it would help?

Wren6991, 2021-03-11:
> On another note, IMO it is quite disturbing that the GPIO Out, In, etc, registers are listed in the SIO section, I understand why but my first reflexes would be to check the GPIO section to learn [...]

There is also this in the description of GPIO functions, two pages after the overview:

[image]

But as you say it is a little hard to find because it's not necessarily what people are looking for. Other people have also struggled to find it, so this new subheading under GPIO will appear in the next docs release:

[image]

kilograham, 2021-04-08:
merged

closed 2021-04-08 by kilograham