raspberrypi/pico-sdk #165: No interrupt function in SDK for the PIO

Volksolive, 2021-02-19:
Sorry in advance if that was already raised.

There isn't any function SDK (or at least in documentation) that handle PIO IRQ whether it is to enable it or clear the IRQ flags.

Also, the description of the IRQ flag in the RP2040 datasheet is IMO not quite clear. I understand that only the 4 lower flags can generate a system IRQ through IRQ0_INTE and IRQ1_INTE but are the 4 lower flag representing the SM number or can SM0 set flag 3 f.ex and IRQx_INTS still shows SM0 as pending? 

yes, I could check it for myself but maybe it is worth having it better explained in datasheet?

kilograham, 2021-06-02:
merged

closed 2021-06-02 by kilograham

lurch, 2021-06-03:
> I understand that only the 4 lower flags can generate a system IRQ through IRQ0_INTE and IRQ1_INTE but are the 4 lower flag representing the SM number or can SM0 set flag 3 f.ex and IRQx_INTS still shows SM0 as pending?

AIUI the `SM0` .. `SM3` field-labels in the "PIO: IRQ0_INTS Register" table in the RP2040 datasheet are slightly misleading, and if SM0 sets flag 3, then bit 11 of IRQ0_INTS would be set. https://github.com/raspberrypi/pico-sdk/pull/406#pullrequestreview-659392052