raspberrypi/pico-sdk #169: Visual Studio Code, fatal error u1033 syntax error ':' unexpected

NicoEichen, 2021-02-19:
hi, when compiling i get this error. 
I'm new to visual studio code. I'm a microcontroller programmer, so i usually write code in c, and pass it to a compiler nad its done, like with arduino ide, or others.
So i wrote a code and got this error:
`[build] CMakeFiles\ControlCortinas.dir\compiler_depend.make(4) : fatal error U1033: error de sintaxis : no se esperaba ':'
[build] Stop.
[build] NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29333\bin\HostX86\x86\nmake.exe"' : c�digo devuelto '0x2'
[build] Stop.
[build] NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.28.29333\bin\HostX86\x86\nmake.exe"' : c�digo devuelto '0x2'
[build] Stop.
[build] Compilación finalizada con el código de salida 2`
 i checked the compiler_depend.make and notice that there are ':' and ' \' where there shouldn't be any :
`CMakeFiles\ControlCortinas.dir\C_\Raspberry_Pico\pico-sdk\src\common\pico_sync\critical_section.c.obj: C\:\Raspberry_Pico\pico-sdk\src\common\pico_sync\critical_section.c` 
like here after .obj there is ':' or between C and : there is a '\'.
other examples:
`C\:\Raspberry_Pico\pico-sdk\src\rp2_common\pico_printf\include\pico\printf.h:`

also in compiler_depend.internal there are similar problems with the '\':
` C\:\Raspberry_Pico\pico-sdk\src\common\pico_sync\critical_section.c`

i'm not a pc programmer, and i don't know much about NMake, or the details of how visual studio code works, etc. so i will appreciate that you can guide me with words like, click here, go to this directory, edit this file, etc. thank you.

lurch, 2021-02-20:
Looks like maybe the `PICO_SDK_PATH` variable hasn't been set correctly? Did you remember to launch VSCode from a Developer Command Prompt instead of just launching it from the Start Menu or Desktop? See section 8.2.4 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf

NicoEichen, 2021-02-20:
that wasn't the problem, after trial and error i notice the problem, when you have already try to compile a program, it will throw this error. Basically after building any program, if you want to edit it you have to delete the files from build folder. I don't know exactly which files to delete. i just deleted everything, close folder in visual studio code and open it again so it can create basic component required in build folder to compile, after that i can compile. 
I hope some1 here can make a workaround this problem so you don't have to delete everything everytime.

Examples folder 1st try to build all:
[examples build all 1st time.txt](https://github.com/raspberrypi/pico-sdk/files/6015150/examples.build.all.1st.time.txt)
Examples folder 2nd try to build all:
[examples build all 2nd time.txt](https://github.com/raspberrypi/pico-sdk/files/6015152/examples.build.all.2nd.time.txt)
Trying to build Hello_Serial after build all:
[Examples Hello_Serial after build all.txt](https://github.com/raspberrypi/pico-sdk/files/6015154/Examples.Hello_Serial.after.build.all.txt)
Deleting everything in build folder and starting from cero, then trying to build hello_serial:
[Examples Hello_Serial before build all.txt](https://github.com/raspberrypi/pico-sdk/files/6015157/Examples.Hello_Serial.before.build.all.txt)
Building blink 1st time:
[Examples blink before build all.txt](https://github.com/raspberrypi/pico-sdk/files/6015158/Examples.blink.before.build.all.txt)
Building blink 2nd time without deleting anything:
[Examples blink 2nd build before build all.txt](https://github.com/raspberrypi/pico-sdk/files/6015160/Examples.blink.2nd.build.before.build.all.txt)

lurch, 2021-02-21:
I just followed section 8.2.4 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf myself in a fresh Windows 10 VM, and I managed to reproduce your problem... and I also found the solution! :grinning: 
It seems there's a bug in the 3.20.0-rc1 version of CMake, which means only a "clean" nmake works, and (as you observed) trying to run nmake a second time results in a failure. I reproduced this using the Developer Command Prompt, so it's definitely an error with CMake rather than with VSCode.

I deleted my `build` folder again, uninstalled CMake 3.20.0-rc1 and then installed CMake 3.19.5 and now I can do multiple builds without any odd errors. I'll add a warning about this to our documentation, thanks for spotting it!

lurch, 2021-02-24:
> I'll add a warning about this to our documentation

This warning is now included in the new PDFs that got published today. I think this issue can be closed @aallan

NicoEichen, 2021-02-24:
Yeah, thank you very much.

closed 2021-02-24 by NicoEichen

aallan, 2021-03-29:
Reported upstream as a CMake bug, https://gitlab.kitware.com/cmake/cmake/-/issues/21997, by @lurch.

lurch, 2021-04-12:
FYI this is now fixed in [CMake 3.20.1](https://cmake.org/download/) :tada: