raspberrypi/pico-sdk #178: Compilation error with -Wall -Wextra -Werror -Wpedantic

tomikaa87, 2021-02-23:
Projects that treats warnings as errors and uses higher warning levels can't be build with the SDK.
There are several issues in the header files, like these (and many more):

```
[build] ../pico-sdk/src/rp2_common/hardware_pio/include/hardware/pio.h:301:17: error: comparison is always true due to limited range of data type [-Werror=type-limits]
[build]   301 |     assert(join >= 0 && join <= 2);
[build]       |                 ^~
```

```
[build] ../pico-sdk/src/rp2_common/hardware_pio/include/hardware/pio.h:358:5: error: missing initializer for field 'pinctrl' of 'pio_sm_config' [-Werror=missing-field-initializers]
[build]   358 |     pio_sm_config c = {0, 0, 0};
[build]       |     ^~~~~~~~~~~~~
```

```
[build] ../pico-sdk/src/rp2_common/hardware_pio/include/hardware/pio.h:104:14: note: 'pinctrl' declared here
[build]   104 |     uint32_t pinctrl;
[build]       |              ^~~~~~~
```

```
[build] ../pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h:143:31: error: comparison is always true due to limited range of data type [-Werror=type-limits]
[build]   143 |     valid_params_if(PWM, mode >= PWM_DIV_FREE_RUNNING && mode <= PWM_DIV_B_FALLING);
[build]       |                               ^~
```

```
[build] ../pico-sdk/src/rp2_common/hardware_vreg/include/hardware/vreg.h:31:25: error: binary constants are a GCC extension [-Werror]
[build]    31 |     VREG_VOLTAGE_1_05 = 0b1010,    ///< 1.05v
[build]       |                         ^~~~~~
```

Since compiling with higher error levels is considered to be a best practice, the SDK should support this case.

lurch, 2021-02-23:
I believe some of these have already been fixed on the `develop` branch, if you'd like to check that out and re-test?

tomikaa87, 2021-02-23:
@lurch Some of them are fixed, but there are still a lot of issues.
I've made a Gist with all the warnings to help locating them: https://gist.github.com/tomikaa87/8a838a2c1608b70c9e4210616c956825

kilograham, 2021-02-23:
We are unlikely to fix -Wpedantic. Do you see any issues without that.

tomikaa87, 2021-02-23:
@kilograham 
There are a few:

```
[build] /opt/homebrew/bin/arm-none-eabi-gcc -DCFG_TUSB_DEBUG=1 -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DCFG_TUSB_OS=OPT_OS_PICO -DPICO_BIT_OPS_PICO=1 -DPICO_BOARD=\"pico\" -DPICO_BOOT2_NAME=\"boot2_w25q080\" -DPICO_BUILD=1 -DPICO_CMAKE_BUILD_TYPE=\"Debug\" -DPICO_COPY_TO_RAM=0 -DPICO_CXX_ENABLE_EXCEPTIONS=0 -DPICO_DIVIDER_HARDWARE=1 -DPICO_DOUBLE_PICO=1 -DPICO_FLOAT_PICO=1 -DPICO_INT64_OPS_PICO=1 -DPICO_MEM_OPS_PICO=1 -DPICO_NO_FLASH=0 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_PRINTF_PICO=1 -DPICO_STDIO_UART=1 -DPICO_STDIO_USB=1 -DPICO_TARGET_NAME=\"pico_bufferless_oled\" -DPICO_USE_BLOCKED_RAM=0 -I../pico-sdk/src/common/pico_stdlib/include -I../pico-sdk/src/rp2_common/hardware_gpio/include -I../pico-sdk/src/common/pico_base/include -Igenerated/pico_base -I../pico-sdk/src/boards/include -I../pico-sdk/src/rp2_common/pico_platform/include -I../pico-sdk/src/rp2040/hardware_regs/include -I../pico-sdk/src/rp2_common/hardware_base/include -I../pico-sdk/src/rp2040/hardware_structs/include -I../pico-sdk/src/rp2_common/hardware_claim/include -I../pico-sdk/src/rp2_common/hardware_sync/include -I../pico-sdk/src/rp2_common/hardware_uart/include -I../pico-sdk/src/rp2_common/hardware_divider/include -I../pico-sdk/src/common/pico_time/include -I../pico-sdk/src/rp2_common/hardware_timer/include -I../pico-sdk/src/common/pico_sync/include -I../pico-sdk/src/common/pico_util/include -I../pico-sdk/src/rp2_common/pico_runtime/include -I../pico-sdk/src/rp2_common/hardware_clocks/include -I../pico-sdk/src/rp2_common/hardware_resets/include -I../pico-sdk/src/rp2_common/hardware_watchdog/include -I../pico-sdk/src/rp2_common/hardware_xosc/include -I../pico-sdk/src/rp2_common/hardware_pll/include -I../pico-sdk/src/rp2_common/hardware_vreg/include -I../pico-sdk/src/rp2_common/hardware_irq/include -I../pico-sdk/src/rp2_common/pico_printf/include -I../pico-sdk/src/rp2_common/pico_bootrom/include -I../pico-sdk/src/common/pico_bit_ops/include -I../pico-sdk/src/common/pico_divider/include -I../pico-sdk/src/rp2_common/pico_double/include -I../pico-sdk/src/rp2_common/pico_int64_ops/include -I../pico-sdk/src/rp2_common/pico_float/include -I../pico-sdk/src/rp2_common/pico_malloc/include -I../pico-sdk/src/common/pico_binary_info/include -I../pico-sdk/src/rp2_common/pico_stdio/include -I../pico-sdk/src/rp2_common/pico_stdio_uart/include -I../pico-sdk/src/rp2_common/pico_stdio_usb/include -I../pico-sdk/lib/tinyusb/src -I../pico-sdk/lib/tinyusb/src/common -I../pico-sdk/lib/tinyusb/hw -I../pico-sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/include -I../pico-sdk/src/rp2_common/hardware_i2c/include -I../pico-sdk/src/rp2_common/hardware_adc/include -I../pico-sdk/src/rp2_common/hardware_dma/include -I../pico-sdk/src/rp2_common/hardware_flash/include -I../pico-sdk/src/rp2_common/hardware_interp/include -I../pico-sdk/src/rp2_common/hardware_pio/include -I../pico-sdk/src/rp2_common/hardware_pwm/include -I../pico-sdk/src/rp2_common/hardware_rtc/include -I../pico-sdk/src/rp2_common/hardware_spi/include -march=armv6-m -mcpu=cortex-m0plus -mthumb -Og -g -Wall -Wextra -Werror -ffunction-sections -fdata-sections -MD -MT CMakeFiles/pico_bufferless_oled.dir/main.c.obj -MF CMakeFiles/pico_bufferless_oled.dir/main.c.obj.d -o CMakeFiles/pico_bufferless_oled.dir/main.c.obj -c ../main.c
[build] In file included from ../pico-sdk/src/common/pico_base/include/pico/assert.h:18,
[build]                  from ../pico-sdk/src/common/pico_base/include/pico.h:20,
[build]                  from ../pico-sdk/src/common/pico_stdlib/include/pico/stdlib.h:10,
[build]                  from ../main.c:5:
[build] ../pico-sdk/src/rp2_common/hardware_pio/include/hardware/pio.h: In function 'sm_config_set_fifo_join':
[build] ../pico-sdk/src/rp2_common/hardware_pio/include/hardware/pio.h:307:17: error: comparison is always true due to limited range of data type [-Werror=type-limits]
[build]   307 |     assert(join >= 0 && join <= 2);
[build]       |                 ^~
[build] ../pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h: In function 'pwm_config_set_clkdiv_mode':
[build] ../pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h:147:31: error: comparison is always true due to limited range of data type [-Werror=type-limits]
[build]   147 |     valid_params_if(PWM, mode >= PWM_DIV_FREE_RUNNING && mode <= PWM_DIV_B_FALLING);
[build]       |                               ^~
```

kilograham, 2021-02-25:
those seem to be as a result of enums being unsigned! did you add a compile flag to do that?

tomikaa87, 2021-02-25:
@kilograham No, I only added those that mentioned in the title.

lurch, 2021-02-25:
#182 has recently been merged which fixes some more warrnings :slightly_smiling_face:  Nice one @kilograham !

albertvanderhorst, 2021-03-11:
A comment. Treating warnings as errors is dangerous. Some practices to get rid of warnings (such as casts) hide problems that need attention. The compiler cannot realize that the double value I put into a single will never overflow, but it will give a warning nonetheless.  That warning shoudl remain to draw attention in a recompilation that there is a possible problem. So "considered best practice" maybe. Considered a "problem caused by bad management" by some people with 50+ years of experience. Unless your management has you in a clinch, the best practice is dutyfully inspect all warnings, head the warnings where justified and ignore them where not.

tomikaa87, 2021-03-18:
@albertvanderhorst 

I agree that if there is a warning, it must be treated properly and avoid hiding it with a cast or something like that.
But why I prefer treating warnings as errors is because if you work on a larger project that uses a CI system and builds for multiple platforms on various compilers (that can differ from the ones used by the developers themselves), these warnings can be lost in the build log, but an error will break the build and this must be investigated by the author of the actual modification. There were cases when the code compiled without issues on the developer's machine on one platform, but the same code thrown warnings on another platform. IMHO treating warnings as errors is a pretty cheap solution to protect us from such problems, and I don't say that this should be the only way to ensure that the code is fine, but it helps.

tomikaa87, 2021-03-18:
> those seem to be as a result of enums being unsigned! did you add a compile flag to do that?

It seems arm-none-eabi-gcc does that by default if there is no enumeration with a negative value. In other compilers (even in arm-linux-eabi-gcc) they are always signed integers.

closed 2021-04-08 by kilograham