raspberrypi/pico-sdk #188: Add pico_status_led library

kilograham, 2021-02-25:
The 1.0.2 release removed the requirement for PICO_DEFAULT_LED_PIN since not all boards have a LED
Equally PICO_DEFAULT_WS2812_PIN was added.

Right now user code has to check for the existence of the pins, set them up etc. We should add a friendly library which is a no-op if none are defined, which also handles output on common pin types (e.g. WS2812).

lurch, 2021-02-25:
> Equally PICO_DEFAULT_WS2812_PIN was added.

Was it? :confused:

hathach, 2021-02-25:
great news, it is worth mentioning that some board also has dedicate pin for neopixel power enable as well.

kilograham, 2021-02-25:
> > Equally PICO_DEFAULT_WS2812_PIN was added.
> 
> Was it? confused
it isn't committed yet

closed 2025-07-28 by kilograham