raspberrypi/pico-sdk #195: Repeated invocation of function "void sleep_us(uint64_t us)" can cause lockup

ascotraces, 2021-02-27:
The attached code based on SDK C/C++ examples code test.c and built under Window10 demonstrates the problem:
[test.zip](https://github.com/raspberrypi/pico-sdk/files/6054267/test.zip)

#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/gpio.h"
#include "pico/binary_info.h"

const uint LED_PIN = 25;

int main() {

   stdio_init_all();

   gpio_init(LED_PIN);
   gpio_set_dir(LED_PIN, GPIO_OUT);
   while (1) {
      gpio_put(LED_PIN, 0);
      sleep_ms(250);
      gpio_put(LED_PIN, 1);
      puts("Hello World\n");
      sleep_ms(250);
	  
	  // invocation of function "void sleep_us(uint64_t us)"
	  // when the number of microseconds to sleep (parameter "us") value
      // is set to 9 program lockup occurs, values of 8 and 10 are OK
	  sleep_us(9);
   }
}

kilograham, 2021-02-27:
attach a UF2

ascotraces, 2021-02-27:
UF2 attached
[test-uf2.zip](https://github.com/raspberrypi/pico-sdk/files/6055283/test-uf2.zip)

ascotraces, 2021-02-27:
[Developer Command Prompt for VS2019.txt](https://github.com/raspberrypi/pico-sdk/files/6055317/Developer.Command.Prompt.for.VS2019.txt)

kilograham, 2021-02-28:
I can see your UF2 hang; sadly i should really have asked for the ELF, but:

Can you tell me output of:

From your pico-sdk and pico-examples directories:
```
git log -n 1
```

Can you send me the ELF that is built, and ideally see if you can reproduce it in debug mode (not sure if you are using VS Code, or cmd line, but you want to add -DCMAKE_BUILD_TYPE=Debug to your initial build setup)

Let me know if a debug build is broken too, if so send me the ELF for that as well

Finally, include the cmake output when starting from a clean/fresh build directory

ascotraces, 2021-02-28:
Git logs, ELF and CMake output for both the original build and a clean build in GitLogsAndtestELF.zip. The clean build also reproduced the hang problem.
[test-uf2.zip](https://github.com/raspberrypi/pico-sdk/files/6057019/test-uf2.zip)

The debug build using -DCMAKE_BUILD_TYPE=Debug works fine. The resultant test.uf2 loads and executes correctly. I have enclosed the ELF and CMake output for information purposes, see file test-debug.zip.
[test-uf2.zip](https://github.com/raspberrypi/pico-sdk/files/6057039/test-uf2.zip)

ascotraces, 2021-02-28:
[GitLogsAndtestELF.zip](https://github.com/raspberrypi/pico-sdk/files/6057074/GitLogsAndtestELF.zip)
[test-debug.zip](https://github.com/raspberrypi/pico-sdk/files/6057101/test-debug.zip)

Ignore the test-uf2.zip files attached in the previous comment. The correct GitLogsAndtestELF.zip and test-debug.zip files are attached to this comment.

lurch, 2021-02-28:
Would you mind trying the same thing with the `develop` branch of pico-sdk, to see if that still has the same problem?

kilograham, 2021-02-28:
ah, useful information would be that you are using stdio_usb!

kilograham, 2021-02-28:
but i can now reproduce the problem (only with gcc 10.2.1)

kilograham, 2021-02-28:
and on the develop branch

ascotraces, 2021-02-28:
I confirm:
a) That if I switch to the develop branch of pico-sdk, then I still has the same problem. 
b) I am using gcc 10.2.1 20201103 (release).
c) I am using usb serial comms.

[GitLogDevelop.zip](https://github.com/raspberrypi/pico-sdk/files/6057744/GitLogDevelop.zip)

lurch, 2021-03-12:
Related PR ( #243 ) has been merged, so I guess this can be closed?

kilograham, 2021-03-13:
merged int 1.1.1 branch

PR #243 (2021-03-10): Timer race condition
fixes #195
closed 2021-03-13 by kilograham