raspberrypi/pico-sdk #200: Build SDK on macOS Catalina

8ga3, 2021-02-28:
Pico SDK 1.0.1

I built the development environment on macOS Catalina.
I got some errors. I had to fix it.

```
diff --git a/src/rp2040/hardware_regs/include/hardware/platform_defs.h b/src/rp2040/hardware_regs/include/hardware/platform_defs.h
index 51c0272..eaac3a5 100644
--- a/src/rp2040/hardware_regs/include/hardware/platform_defs.h
+++ b/src/rp2040/hardware_regs/include/hardware/platform_defs.h
@@ -27,7 +27,7 @@

 // PICO_CONFIG: PICO_STACK_SIZE, Stack Size, min=0x100, default=0x800, advanced=true, group=pico_standard_link
 #ifndef PICO_STACK_SIZE
-#define PICO_STACK_SIZE 0x800u
+#define PICO_STACK_SIZE 0x800
 #endif

 // PICO_CONFIG: PICO_HEAP_SIZE, Heap size to reserve, min=0x100, default=0x800, advanced=true, group=pico_standard_link
diff --git a/src/rp2_common/hardware_irq/include/hardware/irq.h b/src/rp2_common/hardware_irq/include/hardware/irq.h
index 6075118..408f6cb 100644
--- a/src/rp2_common/hardware_irq/include/hardware/irq.h
+++ b/src/rp2_common/hardware_irq/include/hardware/irq.h
@@ -10,7 +10,7 @@
 // These two config items are also used by assembler, so keeping separate
 // PICO_CONFIG: PICO_MAX_SHARED_IRQ_HANDLERS, Maximum Number of shared IRQ handers, default=4, advanced=true, group=hardware_irq
 #ifndef PICO_MAX_SHARED_IRQ_HANDLERS
-#define PICO_MAX_SHARED_IRQ_HANDLERS 4u
+#define PICO_MAX_SHARED_IRQ_HANDLERS 4
 #endif
```
arm-none-eabi-as (2.33.1) could not use u(unsigned) with a immediate data.

```
 // PICO_CONFIG: PICO_DISABLE_SHARED_IRQ_HANDLERS, Disable shared IRQ handers, type=bool, default=0, group=hardware_irq
diff --git a/src/rp2_common/pico_standard_link/new_delete.cpp b/src/rp2_common/pico_standard_link/new_delete.cpp
index ecb04b4..e5fd3b8 100644
--- a/src/rp2_common/pico_standard_link/new_delete.cpp
+++ b/src/rp2_common/pico_standard_link/new_delete.cpp
@@ -17,7 +17,7 @@ void *operator new[](std::size_t n) {
     return std::malloc(n);
 }

-void operator delete(void *p, std::size_t n) noexcept { std::free(p); }
+// void operator delete(void *p, std::size_t n) noexcept { std::free(p); }

 void operator delete(void *p) { std::free(p); }
```

Pico SDK specifies C++11.
This **delete** operator is used in C++14 and above.
It is inconsistent.

```
diff --git a/test/kitchen_sink/CMakeLists.txt b/test/kitchen_sink/CMakeLists.txt
index 18ed84c..f85ed2c 100644
--- a/test/kitchen_sink/CMakeLists.txt
+++ b/test/kitchen_sink/CMakeLists.txt
@@ -50,7 +50,7 @@ target_compile_options(kitchen_sink_options INTERFACE
         -Wextra
         -Wno-unused-parameter
         -Wno-inline
-        -Wnull-dereference
+        #-Wnull-dereference
 #        -pedantic
         -Wall
         -Wcast-qual
```

arm-none-eabi-gcc-5.4.1 cannot use '-Wnull-dereference'.

kilograham, 2021-02-28:
Not sure any off this is a macOS catalina issue, you appear to have a very very very old arm-none-eabi-gcc

8ga3, 2021-02-28:
Getting started with Raspberry Pi Pico
https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf

I installed it according to this document.
I also care that gcc is old.

aallan, 2021-02-28:
> I installed it according to this document.

I wrote that document, and mostly develop on macOS. Instructions have been tested on both Mojave and Catalina.

```
$ brew tap ArmMbed/homebrew-formulae 
$ brew install arm-none-eabi-gcc
```

installs version 9.2.1,

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$
```

not version 5.4.1. 

It's possible your Homebrew installation is old and needs updating, potentially brew installed the latest version of arm-none-eabi-gcc that your currently installed dependences supports. Maybe?

8ga3, 2021-02-28:
@aallan 

Thank you.
The Ardupilot development environment was used.
Gcc 9.2.1 was installed.

```
% brew info ardupilot/px4/gcc-arm-none-eabi
ardupilot/px4/gcc-arm-none-eabi: stable 20160928
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
/usr/local/Cellar/gcc-arm-none-eabi/20160928 (5,502 files, 447.2MB) *
  Built from source on 2020-03-07 at 13:34:58
From: https://github.com/ardupilot/homebrew-px4/blob/HEAD/gcc-arm-none-eabi.rb

% brew info armmbed/formulae/arm-none-eabi-gcc
armmbed/formulae/arm-none-eabi-gcc: stable 9-2019-q4-major
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major (6,009 files, 533.4MB) *
  Built from source on 2021-02-23 at 12:17:56
From: https://github.com/ArmMbed/homebrew-formulae/blob/HEAD/arm-none-eabi-gcc.rb
```

aallan, 2021-02-28:
If  arm-none-eabi-gcc 9.2.1 is installed, why are you asking about problems with arm-none-eabi-gcc 5.4.1?

8ga3, 2021-02-28:
I was unaware that two versions were installed.
cmake used arm-none-eabi-gcc 5.4.1.
This is my mistake.
However, I think I got one insight into forward compatibility.

closed 2021-03-03 by kilograham