raspberrypi/pico-sdk #201: Q: Precedence of FIFO Joining?

soundpaint, 2021-02-28:
Looking at the Pico SDK implementation, I see this code:

static inline void pio_sm_clear_fifos(PIO pio, uint sm) {
    // changing the FIFO join state clears the fifo
    check_sm_param(sm);
    hw_xor_bits(&pio->sm[sm].shiftctrl, PIO_SM0_SHIFTCTRL_FJOIN_RX_BITS);
    hw_xor_bits(&pio->sm[sm].shiftctrl, PIO_SM0_SHIFTCTRL_FJOIN_RX_BITS);
}

That is, if I understand correctly, for clearing the FIFOs, the bit for enlarging RX FIFO is toggled twice, thus restoring it to its original value, while the bit for enlarging TX FIFO is not explicitly changed.
So far, so good.
Now imagine the bit for joining TX FIFO is set when entering this function.
After the call to first hw_xor_bits(), _both_ bits are set, which does not make much sense, since only one FIFO can be enlarged and the other will be set to length 0.  Therefore, I would have expected, that setting the bit for enlarging RX FIFO would implicitly clear the bit for enlarging TX FIFO.  But then, the above code would not restore the TX FIFO bit to its original value.
So, assuming that the above code is correct, really _both_ bits can be set to 1 at the same time?  What will happen in this state?  Is there a precendence of one the two bits over the other one?

Greetings,
Jürgen

lurch, 2021-02-28:
Hmmm, even ignoring the XOR issue mentioned above, I just checked the datasheet and it doesn't specify what happens if you set both `FJOIN_RX` and `FJOIN_TX` to 1 at once.

Wren6991, 2021-03-04:
>  What will happen in this state?

Both FIFOs will read as both full and empty :) It's a useless state, but doesn't do anything undefined, and safely triggers the flush due to the join state changing.

lurch, 2021-03-04:
Ahhh, so it effectively disables the FIFOs?

Wren6991, 2021-03-04:
[image]

Pasting the datasheet listing here. Setting FJOIN_RX ties off the TX FIFO interface. Setting FJOIN_TX ties off the RX FIFO interface. Setting both of them is a safe but useless thing to do. It's maybe unclear from looking at the SDK that there are two one-bit control fields here, not a single two-bit field.

soundpaint, 2021-03-05:
Ok, thank you!  I got it now.  From my perspective, the issue can be closed; but maybe you want to add eventually a clarifying comment in the docs?  :-)

lurch, 2021-03-05:
> maybe you want to add eventually a clarifying comment in the docs? :-)

We've already got an internal issue for that :slightly_smiling_face:

soundpaint, 2021-03-05:
Ok, then I close this issue. :-)

closed 2021-03-05 by soundpaint