raspberrypi/pico-sdk #202: Suggestion: Add check_sm_param(sm);

soundpaint, 2021-02-28:
C SDK functions
  void pio_sm_set_pins(PIO pio, uint sm, uint32_t pins)
  void pio_sm_set_pins_with_mask(PIO pio, uint sm, uint32_t pinvals, uint32_t pin_mask)
  void pio_sm_set_pindirs_with_mask(PIO pio, uint sm, uint32_t pindirs, uint32_t pin_mask)
  void pio_sm_set_consecutive_pindirs(PIO pio, uint sm, uint pin, uint count, bool is_out)
  void pio_sm_drain_tx_fifo(PIO pio, uint sm)
(and possibly more)

have a parameter "uint sm", but -- unlike most other functions with this argument -- no check_sm_param(sm).
Maybe, for consistency, the range check should be added for these functions as well?

lurch, 2021-02-28:
#179 :rofl: :wink:

soundpaint, 2021-03-01:
> but I've not actually tried running any of the output binaries on a Pico.

Me too!  My distributor still says "Pico Sold Out."  I am still waiting to actually get hands on a Pico.  Still, this does not prevent me from already developing stuff for the Pico.  In other words, everything I am posting here, is pure theory and untested.  Maybe I should explicitly add a disclaimer on all of my postings, that I am possibly writing complete nonsense... 😃

soundpaint, 2021-03-01:
Ok, https://github.com/raspberrypi/pico-sdk/pull/179/commits/bf5fd629990651c32e558ebf010c9cd21bd1a9b5 looks promising.  I guess this issue can then be closed...

closed 2021-03-01 by soundpaint