raspberrypi/pico-sdk #210: Q: GPIO FUNCSEL NULL = {0xf or 0x1f}?

soundpaint, 2021-03-01:
* Tables 290 and 292 in the RP datasheet contain the following function categories:
XIP = 0, SPI = 1, UART = 2, I2C = 3, PWM = 4, SIO = 5, PIO0 = 6, PIO1 = 7, GPCK = 8 , USB = 9
(but no "NULL")

* Table 339 mentions additional "31 == NULL" .

* C SDK file gpio.h defines "enum gpio_function" with additional "NULL = 0xf" (in contrast to the above "31 == NULL").

Which is right?

soundpaint, 2021-03-01:
And, by the way, since function gpio_set_function() in gpio.c seems to be (not yet?) implemented, function pio_gpio_init() in pio.h obviously has no effect at all.  Is this behavior subject to be eventually changed, or can I safely ignore any call to function pio_gpio_init()?  Or is there some compiler / linking magic that injects additional code?

lurch, 2021-03-01:
> Table 339 mentions additional "31 == NULL" .
C SDK file gpio.h defines "enum gpio_function" with additional "NULL = 0xf" (in contrast to the above "31 == NULL").
Which is right?

This is only a guess (until one of the hardware engineers comes along and gives us a better answer), but perhaps anything which isn't in the range 0 - 9 is "effectively NULL" ?  (either that, or there's a typo and https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_gpio/include/hardware/gpio.h#L99 was _supposed_ to be `GPIO_FUNC_NULL = 0x1f,` :man_shrugging: )

> And, by the way, since function gpio_set_function() in gpio.c seems to be not (yet?) implemented

Huh? https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_gpio/gpio.c#L28

soundpaint, 2021-03-02:
> > And, by the way, since function gpio_set_function() in gpio.c seems to be not (yet?) implemented
> 
> Huh? https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_gpio/gpio.c#L28

Ah, ok, thanks!  I was erroneously looking at src/host/hardware_gpio/gpio.c, which reads:

  // todo weak or replace? probably weak
  void gpio_set_function(uint gpio, enum gpio_function fn) {
  
  }

lurch, 2021-03-02:
> Ah, ok, thanks! I was erroneously looking at src/host/hardware_gpio/gpio.c, which reads:

It's generally worth ignoring anything in `src/host/` as that's used for building binaries that run on your PC, rather than on the RP2040.

lurch, 2021-04-07:
Thanks for the nudge Graham.
I've now chased this up, and been told "0x1f is the official NULL value" and "anything which isn't 0x1f or in the function table is reserved for future use" so I'll make the necessary changes / PRs :slightly_smiling_face:

lurch, 2021-04-07:
I've reviewed the RP2040 datasheet and the other PDF documentation, and AFAICT there's nothing there that needs changing, it's only `pico-sdk` that needs updating.

kilograham, 2021-04-08:
ah, worth putting fixes #xxx in the PR

closed 2021-04-08 by kilograham

lurch, 2021-04-08:
> ah, worth putting fixes #xxx in the PR desciption

I did :stuck_out_tongue_winking_eye: 
[image]

GitHub seems to be going a bit wonky round the edges lately? :confused:

PR #320 (2021-04-08): Update NULL GPIO function to 0x1f
Fixes #210
commit 4fe995d0 (2021-04-08): Update NULL GPIO function to 0x1f (#320)

Fixes #210