raspberrypi/pico-sdk #211: Ease integration of RTOS, we need cmsis-compatible exceptional handler naming or API to install exceptional handler

xiongyu0523, 2021-03-02:
I'm porting Azure RTOS threadx to PICO and I can't submodule the threadx repo because I need to hook the pendsv and systick exceptional handlers. 

In crt0.S, the SDK declare as isr_pendsv and isr_systick which is private, not conform to CMSIS standard, so the processor port of Azure RTOS need to be modified. 

Or if you can provide a API similar to `irq_set_exclusive_handler` but work for all entries include 0-15 exceptional handlers. Then I can install by user code after main. This is also good.

Now, I can only modify the source of threadx, to align with SDK naming, which is not flexible

kilograham, 2021-03-02:
yes we plan to add ability to set exception handlers (an omission)

also we can make these public; we may not rename them, but at that point you can inject a header via cmake to rename them via #define.

lurch, 2021-03-02:
Links to #21 and #127

michael-ring, 2021-03-02:
Where can I put my vote that at least the cortex specific handlers get proper CMSIS names?

kilograham, 2021-03-02:
Actually since these were private, we can likely rename them.

michael-ring, 2021-03-02:
That would be really great! https://github.com/raspberrypi/pico-sdk/issues/21 has all necessary links to cmsis docs.

xiongyu0523, 2021-03-03:
> yes we plan to add ability to set exception handlers (an omission)
> 
> also we can make these public; we may not rename them, but at that point you can inject a header via cmake to rename them via #define.

So your suggestion is to add below global macro definition in CMakeLists.txt to replace the symbol, right?

```
add_definitions(-DPendSV_Handler=isr_pendsv)
add_definitions(-DSysTick_Handler=isr_systick)
```

kilograham, 2021-03-03:
that was an option that would have workedd - albeit ugly-ly - but i've decided to add the real names now

kilograham, 2021-03-03:
(setting variables yes, via global CMake add_defitions - which is eveil - no!)

xiongyu0523, 2021-03-03:
at least it works and I can now submodule my threadx repo without any change. I left this ticket open and let you decide when a better solution is ready

michael-ring, 2021-03-11:
Could this be a candidate for Release 1.2?

kilograham, 2021-05-06:
merged

kilograham, 2021-05-06:
closing this for the ability to set exception handlers, see #21 if you want the CMSIS exception handler naming instead

closed 2021-05-06 by kilograham