raspberrypi/pico-sdk #213: spi_set_format() docs should explain expected cpol and cpha values

blippy, 2021-03-02:
The docs for spi_set_format() don't explain what values we should give for cpol and cpha, meaning that we have have to examine the source code to find acceptable values. 

spi_cpha_t takes either SPI_CPHA_0 or SPI_CPHA_1 (you can't just specify 0 or 1).

Similarly, spi_cpol_t takes either SPI_CPOL_0 or SPI_CPOL_1.

Please annotate the SDK accordingly.

lurch, 2021-03-03:
https://en.wikipedia.org/wiki/Serial_Peripheral_Interface#Clock_polarity_and_phase ?

blippy, 2021-03-03:
You've missed my point. My point is that you can't just supply the values 0 or 1 for arguments you have to supply SPI_CPHA_0, SPI_CPHA_1, etc.. The only way I found that out was by looking at the header file. I shouldn't need to examine source files to determine what the acceptable values are, it should be in the docs.

lurch, 2021-03-03:
> You've missed my point.

Right, I misunderstood what you were asking, sorry.

> I shouldn't need to examine source files to determine what the acceptable values are, it should be in the docs.

Ahhhh, so I guess what you _actually_ need is for the enums used at https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__spi.html#ga40250e8645b672418cd8d549d0b705d1 to appear in the doxygen output. Yeah, that should be fairly easy to add, thanks for pointing out the omission. :+1:

blippy, 2021-03-03:
Yeah, that's what I'm saying.

lurch, 2021-03-03:
Thanks, fixed by #216 
Please let us know if you spot any others we've forgotten to add to the Doxygen!

lurch, 2021-03-03:
This can be closed now, #216 was merged.

commit 2592c59e (2021-03-03): Add SPI enums to Doxygen output, fixes #213 (#216)

Also fix a few typos
closed 2021-03-05 by kilograham