raspberrypi/pico-sdk #214: Use -g to generate debug symbols in binary, even for "production" images.

WestfW, 2021-03-03:
AFAIK, there is no reason not to use "-g" to generate debug symbols in the binary, even for production images.  This is separate from optimization, and should not result in any differences in the binary.
The current state, which uses "-Og -g" for debug builds, and "-O3" for production, means that the production binary is different than the debug binary, and it's especially annoying to debug (or even look at a disassembled source listing, since you can't generate the intermingled Source/assembly listing without debug symbols, apparently.)

(and... do you want to use `objdump -S `to generate the .dis file instead of "-d"?  that gives you the intermingled source.  Which is frequently pretty scrambled with -O3, but still...

kilograham, 2021-03-03:
`Release` and `RelWithDebInfo` are the same with our without debug info (as set by CMAKE_BUILD_TYPE) or your favorite IDE.

WestfW, 2021-03-03:
RelWithDebInfo seems to build with -O2 rather than -O3, so it still has the "doesn't match Release" problem.

Also, I can't find it, or the other possibilities for CMAKE_BUILD_TYPE documented anywhere...
I guess:

set(CMAKE_BUILD_TYPE "${default_build_type}" CACHE STRING "Choose the type of build, options are: 'Debug', 'Release', 'MinSizeRel', 'RelWithDebInfo'." FORCE)

kilograham, 2021-03-15:
we should make RelWithDebInfo use the same default optimization

kilograham, 2021-08-31:
move to 1.2.1 to consider making RelWithDbgInfo use the same opt level as Release (and indeed adding allowing easy picking of default of O2 vs O3)

closed 2024-08-08 by kilograham